micropython/micropython #4332: nrf ble stack link is dead.

pacmac, 2018-12-02:
https://github.com/micropython/micropython/blob/master/ports/nrf/drivers/bluetooth/download_ble_stack.sh

```
:~ $ wget https://www.nordicsemi.com/eng/nordic/download_resource/45846/3/78153065/80234
--2018-12-02 09:10:43--  https://www.nordicsemi.com/eng/nordic/download_resource/45846/3/78153065/80234
Resolving www.nordicsemi.com... 152.199.39.127
Connecting to www.nordicsemi.com|152.199.39.127|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Cookie coming from www.nordicsemi.com attempted to set domain to nordicsemi-sc-cd-prod-ne-webapp-1.azurewebsites.net
2018-12-02 09:10:45 ERROR 404: Not Found.
```

pacmac, 2018-12-02:
Looks like they have redesigned their website, the files are here:

https://www.nordicsemi.com/Software-and-Tools/Software/nRF5-SDK/Download#infotabs

But it looks like the download of these files can't be automated with direct links, you need to select the sdk version and then select the files you want to download in a single zip, by default the sdk is also included and not just the BT stack files.

glennrub, 2018-12-02:
Hi @pacmac,

I've created a patch for the download links, in PR #4334.
Could you test this one and see if this also works for you?
I'll refine the commit message on that PR a bit if it works for you as well.

dpgeorge, 2018-12-12:
Fixed by 9e5768a6db8cf49e36f773dd97acfa19c187e147

closed 2018-12-12 by dpgeorge