micropython/micropython #4333: nrf build fails on latest - undefined reference to `mp_qstr_frozen_const_pool'

pacmac, 2018-12-02:
```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -rf build-pca10040 
Cleaning up OSX system files...
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LINK build-pca10056-s140/firmware.elf
/tmp/ccg5WZJZ.ltrans7.ltrans.o: In function `qstr_pool_info':
/nfs/qnap/data/uPython/build/micropython/micropython/ports/nrf/../../py/qstr.c:282: undefined reference to `mp_qstr_frozen_const_pool'
/tmp/ccg5WZJZ.ltrans11.ltrans.o: In function `mp_builtin_help.lto_priv.581':
/nfs/qnap/data/uPython/build/micropython/micropython/ports/nrf/../../py/builtinhelp.c:176: undefined reference to `mp_frozen_mpy_names'
/tmp/ccg5WZJZ.ltrans11.ltrans.o: In function `do_load':
/nfs/qnap/data/uPython/build/micropython/micropython/ports/nrf/../../py/builtinimport.c:236: undefined reference to `mp_frozen_mpy_names'
/nfs/qnap/data/uPython/build/micropython/micropython/ports/nrf/../../py/builtinimport.c:236: undefined reference to `mp_frozen_mpy_content'
/tmp/ccg5WZJZ.ltrans11.ltrans.o: In function `mp_import_stat':
/nfs/qnap/data/uPython/build/micropython/micropython/ports/nrf/main.c:272: undefined reference to `mp_frozen_mpy_names'
/tmp/ccg5WZJZ.ltrans12.ltrans.o: In function `mp_micropython_qstr_info.lto_priv.628':
/nfs/qnap/data/uPython/build/micropython/micropython/ports/nrf/../../py/modmicropython.c:105: undefined reference to `mp_qstr_frozen_const_pool'
/tmp/ccg5WZJZ.ltrans19.ltrans.o: In function `do_str':
/nfs/qnap/data/uPython/build/micropython/micropython/ports/nrf/main.c:86: undefined reference to `mp_qstr_frozen_const_pool'
collect2: error: ld returned 1 exit status
Makefile:302: recipe for target 'build-pca10056-s140/firmware.elf' failed
make: *** [build-pca10056-s140/firmware.elf] Error 1
```

dpgeorge, 2018-12-03:
Which board are you trying to build?  Both pca10040 and pca10056 build ok for me.  Note that if you change the use of the `FROZEN_MPY_DIR` option at the command line to make then you must first clean before rebuilding.

takumiando, 2018-12-05:
I got same issue when I built for pca10056 with FROZEN_MPY_DIR option.
Of course, I cleaned before rebuilding.

```
$ python3 ./micropip.py install -p freeze micropython-uasyncio
Installing to: /path/to/micropython/ports/nrf/freeze/
Warning: pypi.org SSL certificate is not validated
Installing micropython-uasyncio 2.1.1 from https://files.pythonhosted.org/packages/0c/33/cbd1ee245621dab9952ebd1d3be840e6acd66e847b33ea88a7f015d2b952/micropython-uasyncio-2.1.1.tar.gz
Installing micropython-uasyncio.core 2.0.1 from https://files.pythonhosted.org/packages/39/9f/619bbe7454419b72073e61b3d9abbc5dfeb54cc4062c2a2d21ad4984db85/micropython-uasyncio.core-2.0.1.tar.gz

$ make BOARD=pca10056 FROZEN_MPY_DIR=freeze
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-pca10056/genhdr

...

CC modules/random/modrandom.c
CC ../../lib/nrfx/mdk/system_nrf52840.c
CC build-pca10056/pins_gen.c
LINK build-pca10056/firmware.elf
build-pca10056/frozen_mpy.c:163:26: error: size of 'mp_qstr_const_pool' differ from the size of original declaration [-Werror=lto-type-mismatch]
 extern const qstr_pool_t mp_qstr_const_pool;
                          ^
../../py/qstr.c:98:19: note: 'mp_qstr_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_const_pool = {
                   ^
../../py/qstr.c:113:26: error: size of 'mp_qstr_frozen_const_pool' differ from the size of original declaration [-Werror=lto-type-mismatch]
 extern const qstr_pool_t MICROPY_QSTR_EXTRA_POOL;
                          ^
build-pca10056/frozen_mpy.c:164:19: note: 'mp_qstr_frozen_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_frozen_const_pool = {
                   ^
lto1: all warnings being treated as errors
lto-wrapper: fatal error: arm-none-eabi-gcc returned 1 exit status
compilation terminated.
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
Makefile:302: recipe for target 'build-pca10056/firmware.elf' failed
make: *** [build-pca10056/firmware.elf] Error 1
```

dpgeorge, 2018-12-05:
Building pca10056 from scratch works for me.  @takumiando can you please try again, without adding any additional files to the freeze/ directory (make sure it just contains test.py), and using the command: `make  BOARD=pca10056 FROZEN_MPY_DIR=freeze V=1`, and then upload/attach the entire build log?

dpgeorge, 2018-12-05:
Thanks for the log.  The problem is that LTO is trying to verify sizes of const data declared in separate source file to check they are correct, but because the qstr_pool_t uses a zero-length array LTO can't do it.  At this point there are two ways I can think of to work around the problem: 1) don't use LTO by building with `make LTO=0` (plus any other options like BOARD); 2) upgrade arm-none-eabi-gcc to a more recent version (8.2.0 is known to work).

@takumiando out of interest, what version of arm-none-eabi-gcc do you use?

takumiando, 2018-12-05:
It works without LTO ;)
>1) don't use LTO by building with make LTO=0

@dpgeorge I'm using this version on ubuntu 18.04 LTS.
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
https://packages.ubuntu.com/en/bionic/gcc-arm-none-eabi

dhalbert, 2018-12-05:
This issue was fixed as of gcc 7.3.1. I submitted a bug report and they fixed it. https://stackoverflow.com/questions/45047888/suppress-wlto-type-mismatch-warning-for-struct-with-flexible-array

dpgeorge, 2018-12-05:
@takumiando thanks for the info, good that it works without LTO.

@dhalbert thanks for letting us know the root cause.  @glennrub should we add something to the nrf/README about LTO, that either it needs a recent version of the compile, or to disable LTO otherwise?

@pacmac I think your build error is different, most likely due to not having a clean build directory to start with.  Can you check if doing a `make clean` fixes it?

glennrub, 2018-12-10:
>  @glennrub should we add something to the nrf/README about LTO, that either it needs a recent version of the compile, or to disable LTO otherwise

@dpgeorge, I support your idea on putting a note of a minimum requirement of 7.3.1 for the nrf-port, as LTO is enabled by default. We have also seen/discussed other issues earlier (nlr_push/nlrthumb), that has the root in LTO issues in the 7.2.1/4Q17 release. Most, if not all, can now be prevented by using the 2Q18 release. I'll prepare some text on this for the readme on the nrf-port.

glennrub, 2019-03-19:
@pacman, @dpgeorge , a section of recommended compilers was merged in e0c6dfe90a3e6265b06265c5b3124c24703776f1. Should we consider to close this issue?

dpgeorge, 2019-03-20:
Yes I think it's fair to close this now.

closed 2019-03-20 by dpgeorge