micropython/micropython #4336: utimeq.push() blindly assumes time value is small int

martijnthe, 2018-12-06:
When using a time value that is larger than a "small int" (tagged pointer) can hold, the `utimeq.push()` blindly assumes the passed value is a small int. In case the time value is not a "small int", the shifted pointer address will be used as time value.

https://github.com/micropython/micropython/blob/master/extmod/modutimeq.c#L132

I ran into this because on Pycom's port, `tick_ms()` end ups up growing bigger than a "small int" (I wonder whether that's really supposed to happen, but that's another story).

cc @pfalcon

martijnthe, 2018-12-06:
@pfalcon thanks for the prompt response.

> ... utime.ticks_ms()/utime.ticks_us(), and those values are guaranteed to be small ints.

I suspected that, reading the comments with the `MICROPY_PY_UTIME_TICKS_PERIOD` definition in `py/mpconfig.h`...

Too bad Pycom's fork is broken on this front :(

cc @robert-hh @iwahdan88

pfalcon, 2018-12-06:
> There're possible documentation improvements, as usual. 

Added explicit guarantee/requirement to my fork's docs: https://pycopy.readthedocs.io/en/latest/library/utime.html?highlight=utime#utime.ticks_ms