micropython/micropython #4340: use https to transfer micropython binaries

ThomasWaldmann, 2018-12-09:
the download page links to http URLs to download the (esp32) binaries.

if one uses the same URL, but with https, it works and is better for authenticity.

so, whatever process updates the links on that page, it should be updates to use https.

https://micropython.org/download#esp32

dpgeorge, 2018-12-10:
Thanks for the report.  This download page has now been changed to use relative URL's for the firmware, so will use https if the page itself is accessed via https.

ThomasWaldmann, 2018-12-10:
Thanks for fixing it for the https site.

So there is just one issue left: the http site does not automatically redirect to the https site, so all downloads there are are, as the site, unencrypted/unauthenticated.

Maybe that should be fixed by a redirect rule, depending on whether it is expected that still anybody would want to specifically use the http site for some reason.

dpgeorge, 2018-12-10:
> Maybe that should be fixed by a redirect rule, depending on whether it is expected that still anybody would want to specifically use the http site for some reason.

I don't want to add a blanket https redirect since I expect some visitors to still use plain http (eg accessing the page from a microcontroller).  I don't know what a good solution here is, but at least there is always the option to go to the https page (and that's the one that can be bookmarked, and linked to from elsewhere).

jonnor, 2024-07-27:
Accessing http://micropython.org/download/ now redirects to the HTTPS version. So this issue can be closed :)

closed 2024-07-27 by robert-hh