micropython/micropython #4344: [STM32] machine.reset_cause() returned on pyb.standby()

rolandvs, 2018-12-10:


dpgeorge, 2018-12-11:
> a wakeup generates a `machine.SOFT_RESET` as result of `machine.reset_cause()`

This is not the intended behaviour.  What exactly is the "wakeup" event that you use to exit standby (hard reset, WDT, RTC WKUP)?  What MCU/board do you use?

rolandvs, 2018-12-11:
Oops, sorry it seems that it was to late yesterday. I tried again with revised script and it blink 5 times which is reset_cause = 4. See script.

```
import machine, pyb
rtc=pyb.RTC()
rtc.init()
rtc.wakeup(None)
blink = machine.reset_cause() + 1
for i in range(blink):
    pyb.LED(3).on(); pyb.delay(10)
    pyb.LED(3).off(); pyb.delay(500)
rtc.wakeup(5000)
pyb.standby()
```

dpgeorge, 2018-12-12:
> which is reset_cause = 4

That is correct, cause 4 is equal to `machine.DEEPSLEEP_RESET`, which is what `pyb.standby()` does.

dpgeorge, 2018-12-29:
It looks like this was buggy for L4 MCUs.  Should be fixed by 4d8504425a1b43a9b90a9a3b7a596e919b8cdb67

closed 2018-12-29 by dpgeorge