micropython/micropython #4350: about license

xiaoxiaohuixxh, 2018-12-13:
Micropython is licensed using the mit protocol. Can I use micropython in my project, but does my project use the apache 2.0 protocol?

xiaoxiaohuixxh, 2018-12-13:
We are deleting the source code of our project that is not the mit protocol, but we want to know if we can change the protocol of our project to apache2.0, because our project contains micropython.

xiaoxiaohuixxh, 2018-12-13:
@pfalcon @joar @suda @swenson

pfalcon, 2018-12-13:
You will need to wait for the reply of the owner of this repository. In the meantime, you can read the license: https://github.com/micropython/micropython/blob/master/LICENSE and/or consult with the license lawyer.

xiaoxiaohuixxh, 2018-12-13:
ths a lot

xiaoxiaohuixxh, 2018-12-13:
I have deleted the source code of the project using gpl protocol, thank you for your reminder!

pfalcon, 2018-12-13:
Well, thanks you too for paying attention to licensing issues, appreciated.

If you'd like to have a builtin editor, there're pure-Python implementations, like https://github.com/robert-hh/Micropython-Editor , https://github.com/pfalcon/pyedit (they might not work fully ideally, but that's the point - the ecosystem needs further development by everyone using it; with enough RAM available, those solutions should be pretty good actually).

xiaoxiaohuixxh, 2018-12-13:
oh，ths a lot，I will pay close attention to these two projects, and we are still working hard to improve the basic library part of micropython. thank you for your help!

dpgeorge, 2018-12-14:
> we want to know if we can change the protocol of our project to apache2.0, because our project contains micropython.

You are free to do what you like with your own source code of your own project, to license it as you see fit.  If you include MicroPython source code then that source code is *licensed to you under the MIT license*.  That means, for these source files that you include, you need to follow what is written in the LICENSE file (or in the individual source files if they have a specific license).  In particular you can "use, copy, modify, merge, publish, distribute, sublicense" the source code, provided "The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software." (please see the full LICENSE text for details).

xiaoxiaohuixxh, 2018-12-15:
ths, i can add apache2.0 license in my code?

dpgeorge, 2018-12-15:
> i can add apache2.0 license in my code?

I didn't look at your code, but if you wrote it you are free to license it however you like.

As I mentioned above, as long as you comply with the MIT license from this repository when using the source code from this repository, you have met your obligations as far as the copyright holders of this code are concerned.  Anything else is out of scope of this project.  Sorry if this sounds formal but there's not much else we can do here.  You may want to ask for advice elsewhere with licensing your project.

closed 2019-01-09 by dpgeorge