micropython/micropython #4354: Failing to import a library.

PeterChudinov, 2018-12-14:
`MicroPython v1.9.2 on 2017-08-23; PYBv1.1 with STM32F405RG` with a 16gb SD card in it.

When I am trying to import the pyCrypto library, I get this error:
`ImportError: no module named 'Crypto.Cipher._AES'`
It's not the only library that I have problems with, here is output of `import rsa`
`ImportError: no module named 'logging'`

To reproduce:
1. Install the library on my computer through `pip3`
2. Copy the library directory
3. Paste it to the board's SD card root
4. `import RSA` in `main.py` or in REPL

andrewleech, 2018-12-14:
pyCrypto is a module largely implemented in C written for desktop CPython, there's a lot more to it than just the python files in the module.

micropython is not compatible with C modules from CPython, as such porting pyCrypto to micropython would require a significant amount of work.

The note about the second time you try to import, yes I've noticed the same behavior. It can be somewhat confusing to be sure.

dpgeorge, 2018-12-14:
> Interesting thing I noted - if you import again, the import pretends to work, but:

That problem is tracked by #967

PeterChudinov, 2018-12-15:
That's quite sad.
Hence, I have to ask: do you know of a way to get RSA and AES encryption work under micropython?

andrewleech, 2018-12-15:
I believe micropython can be built with mbed-tls which appears to support RSA, I haven't looked into that myself though and I doubt the python bindings have been written to expose that in particular.

dpgeorge, 2018-12-17:
> but what about RSA

This is not yet supported.  See #4169 for some proposed initial support.