micropython/micropython #4356: Preparing for next release, v1.10.0

dpgeorge, 2018-12-15:
There have been many commits since v1.9.4 and it's definitely overdue for the next release.  There have been many changes so it's beyond v1.9.5, but the code is not yet ready for v2.0, so v1.10.0 makes sense.

PRs/issues to try and resolve before the release are:
- [ ] docs/library/machine.I2C.rst: Extend writeto to support a list of bufs. #4020
- [ ] DHT22 driver on esp8266 stopped working between v1.9.3 and master #4233
- [ ] esp8266: UART(0) activation for REPL even if expressly deactivated #4314
- [ ] docs: Specify new machine.ADC and machine.ADCBlock classes #4213
- [ ] docs/machine: Specify new class machine.PWM. #4237

peterhinch, 2018-12-15:
Is there any chance of revisiting https://github.com/micropython/micropython/pull/3591?

This is an issue which crops up periodically on the forum. The merit of PyPi is simplicity for new users. Consider a new user of the Pyboard running Windows. To use **upip** they need to create a Linux VM, clone the MicroPython repo and compile the Unix build. These may be major hurdles.

I can't see how to write a single **upip** version which is both highly optimised for ESP8266 and is OS-agnostic. If it can be done, it should be. There is a definite need for _some_ solution to this problem.

hwt9520, 2018-12-15:
i also get the same problem of esp8266, which uart(0) dont works neither with dupterm(None,1) nor dupterm(None)
repl always work when boot esp8266...

dpgeorge, 2018-12-17:
> i also get the same problem of esp8266, which uart(0) dont works neither with dupterm(None,1) nor dupterm(None)
> repl always work when boot esp8266...

@hwt9520 can you please explain in more detail the problem?  What version of the esp8266 firmware did you use, what is the code that you run, and what is it you expect to happen?  If your issue is the same as (or related to) what is described in #4314 then please post the details in that issue.

hwt9520, 2018-12-17:
https://forum.micropython.org/viewtopic.php?f=16&t=5678&sid=c8f4edcd142b1779fe6798127e935d20

mattytrentini, 2018-12-20:
I'd like to see #3658 resolved, even if it's just duplicating the ESP8266 docs and correcting any errors. I can put some time into the docs over the holiday period if that fits in with the 1.10 schedule.

And if I were to pick any single defect to have resolved it would be #4269. It _really_ undermines the quality of the ESP32 port. I know it's not trivial though...

Glad to see a release plan though! Thanks @dpgeorge

peterhinch, 2018-12-20:
I'm looking forward to a release so that bugs fixed long ago (https://github.com/micropython/micropython/issues/3153) find their way into an official release build. I'm not alone in coding round that one.

As for the CoC I don't know how effective it is likely to be in practice. But if it helps towards a greater level of tolerance towards beginners and those whose experience differs from our own I'm all for it. I think skilled people have a duty to share with those younger or less experienced while remaining willing to learn. Someone might be a less skilled Python coder yet have relevant knowledge and experience which they lack.

Just my two penn'orth :)

dpgeorge, 2018-12-20:
Thanks all for the feedback.  I have added the requested 3 issues to the list and will try to get to them.

delivrance, 2018-12-23:
@dpgeorge I guess you are already aware of the Loboris ESP32 port. Is there any chance to bring such features right into this main repo? Most notably: sdcard support, wear leveling and OTA partitions. Or, is there any plan for any other enhancement to the ESP32 port? I see the port is pretty young and it also has nightly builds only (sign that it's still in an early stage).

pfalcon, 2019-01-23:
There're more and more comments appear on tickets/PRs "this should be addressed after 1.10". But what's being done to make that release? There're no comments here for a while. And the list contains strange items like:

> top: Add CODE_OF_CONDUCT.md document from www.contributor-covenant.org. #4358

How adding code of conduct relates to the release? CoC deals with community interaction, while a release is well, a release. Adopting CoC in non-fiat way usually requires collecting community feedback, which requires time. Making that process deadlock the release is again, strange.

pfalcon, 2019-01-23:
Oh, and btw, the release should be good enough reason to get response to https://github.com/micropython/micropython/pull/4363.

dpgeorge, 2019-01-25:
> But what's being done to make that release? There're no comments here for a while

You can see that the list at the top is being updated with items as they are completed (and discussion happens in those issues/PRs).

----

Now that a majority of the items are done I'm happy to make the release ASAP, in the interest of moving forward.

Following earlier release numbering, the next one will be v1.10 and not v1.10.0 as previously stated.

dpgeorge, 2019-01-25:
Builds have all been checked, relevant tests run, and release notes are being prepared.

rlnktt, 2019-01-25:
👍 Waiting eagerly @dpgeorge

dpgeorge, 2019-01-25:
v1.10 is tagged at 3e25d611ef3185b68558a20057d50b0d18dc67a0

dpgeorge, 2019-01-25:
Remaining items were done: GitHub release made, esp-extra-scripts branch rebased and force-pushed, micropython.org/download updated (ordering of some firmware is funny but that will go away in time), and docs updated.

closed 2019-01-25 by dpgeorge