micropython/micropython #4361: __base__ attribute for class is not implemented

nanfang2000, 2018-12-15:
I have a class Application which inherit from class View:
```
class Application(View):
    ...
```
I want to get base class (View) of the class Application by this way:
`Application.__base__`
But I got error "AttributeError: type object 'Application' has no attribute '__bases__'"

Well this code works in CPython3

nanfang2000, 2018-12-16:
My fix:
in line 1044, function mp_load_method_maybe(), runtime.c:
```
} else if (attr == MP_QSTR___base__) {
        mp_obj_type_t* t = MP_OBJ_TO_PTR(obj);
        if (mp_obj_is_instance_type(t))
        {
             dest[0] = MP_OBJ_FROM_PTR(t->parent);
        }
```
It seems works as expected.

pmp-p, 2018-12-16:
could be usefull, need #include "py/objtype.h" too at the top

pmp-p, 2018-12-16:
@nanfang2000, you can edit the file in the browser, then click pullrequest to do like that :
https://github.com/micropython/micropython/compare/master...pmp-p:patch-3

nanfang2000, 2018-12-19:
Thanks Paul for verifying it.
Here is my pull request :
[https://github.com/micropython/micropython/pull/4368](https://github.com/micropython/micropython/pull/4368)

jonnor, 2024-09-29:
__bases__ was implemented and can be used for this purpose, by doing __bases__[0] in the case there is a single base-class. See discussion in https://github.com/micropython/micropython/pull/4368 for why this was preferred over __base__.

closed 2024-09-29 by jonnor