micropython/micropython #4365: sys.exit for esp32?

gengshenghong, 2018-12-18:
what is the behaviour of sys.exit for esp32 now? Is there a way to "reset" micropython task without resetting whole esp32. I want a way to exit micropython without reset esp32. I think sys.exit is the correct answer, but it seems not work on esp32 (nothing happened?). Any reason sys.exit does not cause "soft reset" on esp32?

It seems ‘ctrl+d' in repl mode will do "soft reset" which is what I want. But how to do that using python code? I want to upload a script to esp32 (which may be written by user, not me, it may have some loops or threads for example) and want to reset micropython to re-run user script (again, as I do not know what has been written, I cannot "exit"/"kill" the previous one). But I do not want to reset whole esp32 as I'll add code to run some other tasks together with "mp_task", which should not be restarted.

Please share your ideas if this can be done on esp32.

Thanks,
Shenghong

dpgeorge, 2018-12-20:
> what is the behaviour of sys.exit for esp32

`sys.exit()` is equivalent to `raise SystemExit` and should behave the same on all bare metal targets (esp8266, esp32, stm32, nrf): it stops the current script and drops to the REPL.

> It seems ‘ctrl+d' in repl mode will do "soft reset" which is what I want. But how to do that using python code?

`sys.exit()` did originally do a soft reset (same as ctrl-D) but that was changed a while ago to the above behaviour.  In its place it was proposed to add `machine.soft_reset()` and there is a discussion about this in #2508.  So the action here would be to agree on the behaviour of `machine.soft_reset()` and implement it in esp32 and other ports.

gengshenghong, 2018-12-20:
@dpgeorge Thank you for your clarification. Yes, I agree we should have `machiine.soft_reset()`. I have merged code from  https://github.com/micropython/micropython/commit/57226a2b7f91bd53f0bf392de713be937ede3f5f
 into my local esp32 ports, and confirmed it fulfilled my requirements. 

Thanks,
Shenghong

closed 2018-12-20 by gengshenghong

rolandvs, 2018-12-20:
@dpgeorge Would it be something to give sys.exit() an argument, so argument () or (0) would be "normal" and drop back in REPL, but something else forces a full reset of the system.... (STM32).

dpgeorge, 2018-12-21:
> Would it be something to give sys.exit() an argument, so argument () or (0) would be "normal" and drop back in REPL, but something else forces a full reset of the system

That won't work because sys.exit() and raise SystemExit already take an optional argument which is the exit error code.