micropython/micropython #4370: "array" module should be "uarray" instead

pfalcon, 2018-12-20:
This is one of the older inconsistencies. The original idea was that array module is pretty focuses and thing, so it would be possible to implement it almost completely under the original name. Well "almost completely" is not completely, CPython's "array" has some historical baggage which absolutely makes no sense to add to uPy's "array" (because it's superseded by CPython's own buffer protocol for example), and yet some 3rd-party software may use it. So, following the uPy best practices of ensuring extensibility would be really helpful.

So, better late than never. Propose course of action: make "uarray" the official name, while keep "array" alias for the time being. Update docs to mention only "uarray", make announcement that software targeted for MicroPython should use uarray.

dpgeorge, 2019-01-17:
For consistency within uPy I agree it does make sense to rename it to `uarray`.  But then the same question should be asked of the remaining modules: `math`, `cmath`, `sys`, `gc`.  I don't think any of those fully implement the corresponding CPython module, so by the same argument they should also be renamed.

> make announcement that software targeted for MicroPython should use uarray.

Software should target Python, and MicroPython aims to provide a compatible implementation.  So there can be weak links provided for users to continue to use `array` as normal.

pfalcon, 2019-01-23:
> But then the same question should be asked of the remaining modules: math, cmath, sys, gc.

Of these, sys and gc are treated as "system" modules, i.e. which should not be used "too much" by application code (e.g. perhaps could be used in localized place on startup, not as means to function).

math/cmath are supposed to be(come) complete, I assume that's the reasoning behind adding math.factorial() which is otherwise rather useless.

I don't think it's realistic for "sys" to ever become "usys". More realistic is to allow to override/add stuff there, like `builtins`. If otherwise you're ok to make gc/math/cmath into ugc/umath/ucmath, well - only better. "array" is still the first candidate, as it really miss a bunch function (which are adhoc and legacy, so never should be added to MicroPython).

dpgeorge, 2019-01-24:
> "array" is still the first candidate, as it really miss a bunch function

AFAIK it's just `array.ArrayType` (which is an alias of `array.array`) and `array.typecodes`, or do I miss something?

pfalcon, 2019-01-24:
> or do I miss something?

I personally meant a bunch of adhoc .from*() methods: https://docs.python.org/3/library/array.html#array.array.frombytes . Those are totally superfluous in the presence of generic buffer protocol. So no need to add them to MicroPython's array module, ever. But then it stays incompatible with CPython's module, and makes sense to give users a chance to implement them in a wrapper, by freeing "array" namespace and moving to "uarray".

dpgeorge, 2019-10-22:
`array` was renamed to `uarray` in 21a60935a5d5885817881b4f690fdfd3ef009100, docs updated in a2eea57b1d5456696598703aa4ffdbc7e9fb52ea

closed 2019-10-22 by dpgeorge