micropython/micropython #4376: RFC: Split compile-time vs runtime memory allocation

pfalcon, 2018-12-23:
Currently, any attempt to play with alternative GCs is hampered by the fact that lexer/parser/compiler/emitter (collectively named "compiler") share the heap with runtime. And as compiler data structures are multiple and complex, they quite influence the runtime heap, making it hard to track operations there or add changes to it step by step.

Thus, it would be nice to decouple compile-time allocation from runtime.

pfalcon, 2018-12-23:
@dpgeorge, Let me know if you're interested to address this.

dpgeorge, 2022-06-24:
Using a relatively small value for `gc.threshold()`, eg 5000, actually does pretty well here, to compress the heap during initial imports.

Also, CircuitPython did make GC improvements to separate long-lived objects from short-lived ones.  See #3586.  I'll close this issue in favour of that one.