micropython/micropython #4378: Weirdness with code_info_size field of .mpy bytecode

pfalcon, 2018-12-25:
So, we have https://github.com/micropython/micropython/blob/master/py/bc.h#L41 which says "code_info_size  : var uint |    code_info_size counts bytes in this chunk". Well, that's ambiguous, but common sense is that length field itself is not included in the length it specifies, **especially** if this length field is included as varlen encoding. And yet https://github.com/micropython/micropython/blob/master/py/showbc.c#L94 has a code where length of the field *is* included in the length it encodes.

Why this acrobatics? Do we have a test for what happens when the length jumps from 0x7f straight into 0x81? I don't see any special handling in https://github.com/micropython/micropython/blob/master/py/emitbc.c#L345 .

pfalcon, 2019-01-10:
> The variable code_info_size field includes the number of bytes in this field itself. 

Thanks for confirming this.

> There is a TODO in emitbc.c that this could be improved.

I assume this comment is:

~~~
    // Write size of the rest of the code info.  We don't know how big this
    // variable uint will be on the MP_PASS_CODE_SIZE pass so we reserve 2 bytes
    // for it and hope that is enough!  TODO assert this or something.
~~~

> Actually, there are always 2 bytes reserved for this field in the code-info chunk. If only 1 byte is needed then there is a 0x00 padding byte at the end of the chunk. If more than 2 bytes are needed then it will probably fail, but such a case requires a very large input script (mayb 16k lines or more).

So the situation looks like:

* Initially, this field was chosen to be varint by usual reasons: save bytes for small functions, while be able to accommodate longer (and just long) functions.
* But implementation was written to assume max 2 bytes in it.
* Also, turned out that it's hard to calculate number of bytes needed to encoded it (1 or 2), so initially, 2 bytes is reserved for it. But if in the end it happens to take just 1 byte, we can't recover an extra byte (we already calculated size of everything), so we have to add a padding byte at the end of structure. In other word, we chose varint encoding, spend extra time encoding/decoding it, and in the end just workaround it variable-sizedness, for everything to work like fixed-size.
* As a final touch, value encoding in the varint field includes the (variable) size of the field itself, which only complicates pre-calculation of its size.

It's easy to conclude that something is worth changing here, except perhaps more than one thing needs to be changed in the bytecode encoding, starting with requirements of both speeding up function calls and encoding size.

dpgeorge, 2019-10-01:
Fixed by c8c0fd4ca39fbdcf9ca5f2fc99ca4d6b81a28b65

closed 2019-10-01 by dpgeorge