micropython/micropython #4379: [ESP32] machine.ADC should be able to be initialised with pin numbers as well as machine.Pin

mattytrentini, 2018-12-26:
See the [Documentation for ESP32](https://forum.micropython.org/viewtopic.php?f=18&t=5347#p32747) forum post. It appears that for at least the PyBoard/STM32 and ESP8266 ports `machine.ADC` can be initialized by either a pin number (int) or from an instance of `machine.Pin`. The ESP32 port only allows an instance of `machine.Pin` for ADC init.

I suggest we allow the ESP32 port to also accept pin numbers:

``` python
>>> import machine
>>> a = machine.ADC(35) # Currently fails. Instead must use:
>>> a = machine.ADC(machine.Pin(35))
```

_Either_ of those last two lines should operate correctly.

(Incidentally I _prefer_ the ESP32 behaviour of only accepting `machine.Pin` but consistency is more important in this case.)

Note that there are efforts to overhaul the ADC interface (See #3943 and #4213). The proposed change above could be introduced in a shorter time frame, maybe even for v1.10 (see #4356) and ought to complement the upcoming overhaul.

pfalcon, 2018-12-26:
> (Incidentally I prefer the ESP32 behaviour of only accepting machine.Pin but consistency is more important in this case.)

So, consistency is accepting a Pin instance everywhere (beyond ADC too). Anything else is port-specific/legacy/undocumented behavior.

mattytrentini, 2018-12-27:
Just to clarify @pfalcon, are you agreeing with the suggested change?

If so I'll get on to making the change...

pfalcon, 2018-12-27:
I don't. For classes like UART, I2C, SPI, they *must* take a pin object, because any of them can be implemented in bitbanging manner on absolutely any Pin, e.g. you can do SPI over an I2C GPIO extender connected to a UART port.

Then, consistency calls to use Pin objects to refer to pins everywhere in `machine` API, even if freedom of which pins to use with other blocks is not so pervasive. (But there's nothing wrong to have an ADC extender over SPI, so it's just the same actually).

mattytrentini, 2018-12-28:
I'm really not too concerned whether we do or don't allow pins to be specified by integer for init but I do want _consistency between ports_. If you think it should only be possible to init ADC by passing in a Pin object then the following needs to be done:

1. Modify the ESP8266 port to accept Pin objects and not pin integers
    - Currently only `machine.ADC(0)` works, not `machine.ADC(machine.Pin(0))`
    - This is a _major breaking change_ for ESP8266 - _all_ existing code will fail
2. Documentation needs to be updated
    - Currently some of the documentation examples use pin integers
3. PyBoard needs to be considered
    - It _seems_ to also allow the string syntax to specify the pin, not sure if it accepts `machine.Pin`
4. Need to update the ADC overhaul proposals (#3943 and #4213)
    - It's stated there that pin integers can be used for init

They're not changes we should make lightly since it will break a lot of existing code...

mattytrentini, 2018-12-28:
Also worth noting [#33](https://github.com/micropython/micropython-esp32/issues/33) from the old ESP32 repo where the `ADC(Pin())` syntax is noted.

pfalcon, 2018-12-28:
So, I'm explaining on what principles the original hardware API was designed - simplicity and consistency. In general case, pin identifiers are not "numbers". They can be:

~~~
my_pin = machine.Pin(BLOCK_1, PORT_A, PIN_30, gpio_security_manager(credentials))
~~~

You can't pass all those arbitrary params everywhere, and it's the idea that pins are instantiated in one place, represented by one type of object, Pin, and all other parts of API use that object to access pins.

Of course, one can argue that ADC pins are different from normal digital pins, that they are actually "channels" and not pins, etc., etc. But the original idea of hardware API remains the same, the above.

> consistency between ports

Just a note that if you look closer, you'll find dozens, maybe hundreds of inconsistencies around. Not all of them need to be fixed (one reason is obvious - it may break existing stuff). But neither it means that old inconsistencies forward. So for example, if old port accepted raw numbers for ADC channels, it doesn't mean that all new ports are forever doomed to do the same.

Good points on the docs and the current proposals to finalize the ADC specs. The latter would be indeed the place to which to apply the effort now to clarify the situation.

jonnor, 2024-09-29:
It sounds like consensus was to prefer to have Pin() going forward. And to rather let older ports also support Pin specifications. All the referenced PRs which enable have been merged. So I think this is a wont-fix?