micropython/micropython #4380: Disappointing float precision

GustavoAle, 2018-12-27:
Everything in MicroPython is incredible, except for the float precision. I care a lot about float values precision, and MicroPython offers only 5-6 decimal places precise! In the other hand giant numbers like `500!` can be calculated easily with 100% precision..

At first I thought it was a problem with print, but I performed logical tests and the problem seems to be precision, examples:
```
>>> import math
>>> x = math.pi
>>> x > 3.141592
False
>>> y = math.sqrt(2)
>>> y > 1.4142135
False
```
Both tests should return true. 

I also tried to approximate _e_:
```
>>> e = eulercalc()
>>> e
2.71828
>>> e >= 2.718281
False
```
The same `eulercalc()` function in Linux's python returns `2.7182818284590455`

Is there anything that can be done about it?

peterhinch, 2018-12-27:
The factorial example uses integer arithmetic. This supports arbitrary precision in both CPython and MicroPython. Both deliver the same answer for 500! - albeit as math.fact() on CPython and explicit code on the 32-bit hardware I have to hand.

Single precision MicroPython floats have 21 bits mantissa, 8 bits exponent plus a sign bit.

GustavoAle, 2018-12-28:
21 bits for mantissa is a low value in my perspective, I'm currently working with STM32 and ESP based boards so there are not double precision firmware releases. Now I'm figuring how to implement it by myself. May someone can help me at this task?

robert-hh, 2018-12-28:
Once you have set up the build environment, it's easy for PyBoard aka STM32. Just call:

make MICROPY_FLOAT_IMPL=double

Trying that for ESP32 fails (mpconfigport.h has to be changed here). It results in an linker error.

dpgeorge, 2019-01-17:
With commit eb446ec2276baa7fa1d11056df39de1143487c06 double precision floats can now be use on the esp32 port.  Simply change the relevant line in ports/esp32/mpconfigport.h to:
```c
#define MICROPY_FLOAT_IMPL                  (MICROPY_FLOAT_IMPL_DOUBLE)
```

closed 2019-01-17 by dpgeorge