micropython/micropython #4381: Problem module "re" ESP32

AlexanderSemenov1, 2018-12-27:
What have I done wrong?
In python 3 it works!

>>> import re
>>> re.match(r'...', 'asd').group()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 2 positional arguments but 1 were given

AlexanderSemenov1, 2018-12-27:
Need to do so:   .group(0) ,  happened by accident

>>> import re
>>> re.match(r'...', 'abcd').group(0)
'abc'
>>> 

Documentation on the site http://micropython.org  with a very bad description, no examples.

Jongy, 2019-02-07:
It's confusing because the docs show the function as `.. method:: match.group([index])`, while `index` is actually a required parameter.

The module itself can be changed to match CPython's `re` module, but I think it's one of the spots we can retain MicroPython simpler. So I'll just modify the docs.

dpgeorge, 2019-02-08:
Documentation was updated in a0d60c574a96bb873b0e6822d009031fd5fa3a77

closed 2019-02-08 by dpgeorge