micropython/micropython #4383: ESP32 4th UART issue

hetvishah08, 2018-12-28:
I am trying to develop an application using ESP32 DevkitC that requires 4 UART ports.I know that UART(0) is disabled for REPL and this leaves me with only 2 UART ports. I want to use GPIO4 and GPIO5 as my 4th UART port. I am not able to create a new UART port as it says:

Traceback (most recent call last):
  File "main.py", line 5, in <module>
ValueError: UART(3) does not exist
MicroPython v1.9.4-761-g7cd59c5bc on 2018-12-27; ESP32 module with ESP32
Type "help()" for more information.
>>> 

On going through machine_uart.c  i found out these lines:

    if (uart_num < 0 || uart_num >= UART_NUM_MAX) {
        nlr_raise(mp_obj_new_exception_msg_varg(&mp_type_ValueError, "UART(%d) does not exist", uart_num)); }
  

I think by changing the variable UART_NUM_MAX value to 3,I can add one more UART port but i couldn't find this variable.Where is this variable assigned?
Also if there is any other method to add 4th UART please suggest me.

robert-hh, 2018-12-28:
It is defined in esp-idf/components/driver/include/driver/uart.h. The way it is defined it hardly can be changed.

hetvishah08, 2019-01-02:
Thank you @robert-hh. Any way other than changing UART_NUM_MAX, I could use 4 serial ports using micropython.Like we have Software Serial library for Arduino?

robert-hh, 2019-01-02:
The ESp32 has three UARTs. I do not know of  an implementation for a software UART. For short messages that may be possible using the RMT module, which can create precisely timed bit streams.

dpgeorge, 2019-01-09:
As mentioned, the esp32 only has 3 hardware UARTs, so this is a limitation of the hardware.

closed 2019-01-09 by dpgeorge