micropython/micropython #4387: STM32 BOR not enabled. Flash may be unreliable below 2.7V

chrismas9, 2018-12-30:
STM32 port does not set `BOR_LEV`. This means brown out reset is disabled and the STM32 is only reset when supply drops below ~1.7 V (Vpdr threshold). During a brownout between 2.7 V and 1.7 V the flash memory may be unreliable because the latency (wait states) are too low for low voltage operation.

Reset threshold has been verified on a NUCLEO board where it is possible to power the MCU separately from the STLINK REPL port. When MCU supply is dropped below 1.7 V and raised the reset cause is PWRON_RESET.

One solution would be to add BOR_LEV to mpconfigboard.h with a default of BOR_LEVEL3 and list the four BOR values, voltage ranges and corresponding flash latency values. This would not impact any STM32 running from USB, 5V, 3.3V or LiPo (4.2 to 3 V) but would require a custom build for 2 x AA cell or 1.8 V.

If this is implemented should BROWNOUT be added as a machine.reset_cause() type?

dpgeorge, 2019-01-16:
> One solution would be to add BOR_LEV to mpconfigboard.h with a default of BOR_LEVEL3 and list the four BOR values, voltage ranges and corresponding flash latency values.

It seems that the only way to set the BOR_LEV is by writing to the flash, and clearing the state can only be done via a mass erase.  The setting is therefore non-volatile.  So I think it's better to configure this BOR_LEV outside the firmware, eg during DFU programming, or via a special user-created MicroPython script which prepares a board for use (and could write other items to the OTP region).

For the latter case (being able to set it from MicroPython) it would require some code to directly write flash areas.  This can be done using direct STM registers, but it might be nice to provide a simple class or methods to do this (eg stm.flash_write, stm.otp_write).

dpgeorge, 2019-01-24:
As an example of some code which can write the option bytes:
```python
import stm

OPTION_F4 = 0x1fffc000
OPTION_F7 = 0x1fff0000
OPTKEY1 = 0x08192a3b
OPTKEY2 = 0x4c5d6e7f

def read_option(option_mem):
    o1 = stm.mem32[option_mem]
    o2 = stm.mem32[option_mem + 8]
    return o1 & 0xffff | (o2 & 0xffff) << 16

def write_option(value):
    # unlock
    stm.mem32[stm.FLASH + stm.FLASH_OPTKEYR] = OPTKEY1
    stm.mem32[stm.FLASH + stm.FLASH_OPTKEYR] = OPTKEY2
    # check BSY in FLASH_SR
    while stm.mem32[stm.FLASH + stm.FLASH_SR] & (1 << 16):
        pass
    # write value to FLASH_OPTCR
    stm.mem32[stm.FLASH + stm.FLASH_OPTCR] = value
    # set OPTSTRT in FLASH_OPTCR
    stm.mem32[stm.FLASH + stm.FLASH_OPTCR] |= 1 << 1
    # wait for BSY to clear
    while stm.mem32[stm.FLASH + stm.FLASH_SR] & (1 << 16):
        pass
    # lock
    stm.mem32[stm.FLASH + stm.FLASH_OPTCR] |= 1
```

WARNING: be very careful using the `write_option()` function because it may completely lock the flash if passed in the wrong value.

To enable BOR (on an STM32F4xx) do something like this:
```python
# read initial options
o = read_option(OPTION_F4)
print(hex(o))

# set level 3 BOR
o = o & ~0xc | 0x00

# change the options
write_option(o)

# read new options
o = read_option(OPTION_F4)
print(hex(o))
```

dpgeorge, 2019-06-24:
As mentioned above, BOR functionality is best set up by the user when preparing a board for use.

closed 2019-06-24 by dpgeorge