micropython/micropython #4388: Error on boot [ESP32]

karseyr, 2019-01-01:
Whenever my ESP32 device resets, it outputs a bunch of the usual starting info.  However, it seems to also cause errors.  Maybe this is something that can be fixed?  

```
[0;32mI (347) cpu_start: Pro cpu up.[0m
[0;32mI (347) cpu_start: Single core mode[0m
[0;32mI (347) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (351) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (357) heap_init: At 3FFC0E00 len 0001F200 (124 KiB): DRAM[0m
[0;32mI (363) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (369) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (376) heap_init: At 400915E0 len 0000EA20 (58 KiB): IRAM[0m
[0;32mI (382) cpu_start: Pro cpu start user code[0m
[0;32mI (175) cpu_start: Starting scheduler on PRO CPU.[0m
Traceback (most recent call last):
  File "_boot.py", line 10, in <module>
  File "inisetup.py", line 28, in setup
  File "inisetup.py", line 6, in check_bootsec
  File "flashbdev.py", line 13, in readblocks
OSError: [Errno 5] EIO
OSError: [Errno 1] EPERM
OSError: [Errno 1] EPERM
MicroPython v1.9.4-771-gb33f108cd on 2018-12-31; ESP32 module with ESP32
Type "help()" for more information.
```

karseyr, 2019-01-01:
I also get errors trying to set up WebREPL...  What on earth is wrong with this "Stable" firmware??

`>>> import webrepl_setup
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "webrepl_setup.py", line 102, in <module>
  File "webrepl_setup.py", line 71, in main
  File "webrepl_setup.py", line 40, in get_daemon_status
OSError: [Errno 1] EPERM
`

karseyr, 2019-01-01:
It seems like my problem was very simple to fix: 
I re-flashed the .bin file with the graphical esptool from [https://www.espressif.com/en/support/download/other-tools]here(url), but I ticked the "DoNotChangeBin" box, and now the boot.bin file is included in my install, and I get fewer errors :)

robert-hh, 2019-01-01:
maybe you flash is corrupted. Try erasing the flash and re-installing the firmware. Most of the debug messages after boot.py is run can be turned off with:
```
import esp
esp.osdebug(None)
```

karseyr, 2019-01-01:
@robert-hh I think it's actually fine now, I just forgot to close this.  As a note to future viewers: don't let the ESP Flashing Tool change your binaries when you flash them!

closed 2019-01-01 by karseyr