micropython/micropython #4389: Necessity for Serial flush!

cemox, 2019-01-01:
This is rather recommendation than issue. The code below sends a string message from UART1 and is supposed to pull the output pin LOW :

`mypin.on()`
`uart1.write("a long sentence")`
`mypin.off()`

when I checked the signal (part tx and output) with logic analyzer, I saw that pin is pulled low long before the uart transmission has been completed. A function like `uart.flash()` which checks the relevant uart status bits for completion of the transmission would be very handy.

cemox, 2019-01-02:
By the way, I solved it by making a modification to machine_uart.c file as this : I added internal function call uart_flush(self->uart_id); into uart.write function in machine_uart.c file as shown below:
`STATIC mp_uint_t pyb_uart_write(mp_obj_t self_in, const void *buf_in, mp_uint_t size, int *errcode) {
    pyb_uart_obj_t *self = MP_OBJ_TO_PTR(self_in);
    const byte *buf = buf_in;

    /* TODO implement non-blocking
    // wait to be able to write the first character
    if (!uart_tx_wait(self, timeout)) {
        *errcode = EAGAIN;
        return MP_STREAM_ERROR;
    }
    */

    // write the data
    for (size_t i = 0; i < size; ++i) {
        uart_tx_one_char(self->uart_id, *buf++);
    }

    // added by cemox, wait for completion of transmission
    uart_flush(self->uart_id);

    // return number of bytes written
    return size; 
}`

dpgeorge, 2019-01-09:
I assume you are working with an esp32 board?  In that case there is a transmit buffer on the UART with a default size of 256 characters.  You can configure this buffer to be 0 size which will disable this buffering behaviour, and `uart.write()` will only return once all characters are written.  Eg: `uart.init(txbuf=0)`.

cemox, 2019-01-09:
I am sorry, I forgot to mention that I am using ESP8266 board which has two uarts (one is only tx capable). Micropython is great, but using uart communication is very difficult. Thanks for the comment.

dpgeorge, 2019-01-10:
> I forgot to mention that I am using ESP8266 board

Ok, thanks for the clarification.  Indeed the esp8266 also has a hardware UART tx buffer (126 bytes it seems).  I agree that it's important to be able to control this tx buffer in some way, either to disable it, flush it, and/or check what level it is at.

pmp-p, 2019-01-25:
indeed it would be nice to have sys.stdout.flush() / sys.stderr.flush() whenever it's possible

pmp-p, 2019-07-06:
also if aiming toward 3.4 compatibility , then after 3.3 "flush"  keyword was added to print
https://docs.python.org/3/library/functions.html#print