micropython/micropython #4390: NodeMCU v1.0 stuck at deep sleep rtc wakeup

misan128, 2019-01-01:
Following the docs i tried the deep sleep example
`import machine

# configure RTC.ALARM0 to be able to wake the device
rtc = machine.RTC()
rtc.irq(trigger=rtc.ALARM0, wake=machine.DEEPSLEEP)

# check if the device woke from a deep sleep
if machine.reset_cause() == machine.DEEPSLEEP_RESET:
    print('woke from a deep sleep')

# set RTC.ALARM0 to fire after 10 seconds (waking the device)
rtc.alarm(rtc.ALARM0, 10000)

# put the device to sleep
machine.deepsleep()`

but when it wakes up it gets stuck in booting and doesn't let me erase the flash memmory or anything cause everytime i reset the device it keeps "waking up" from deep sleep and it enters to it once again.

I have tried the esptool.py to erase memory but it doesn't find the device with the usbserial converter nor usb.

PikWay, 2019-01-02:
You've bricked it. I've also managed to do this couple times in past. 
I've managed to erase flash with esptool, but you need to be fast: 
1) Connect your nodeMCU 
2) prepare esptool cmd in terminal/cmd 
3) Press reset button on nodeMCU and exec #2 almost the same time. 

If failed - repeat #3.

robert-hh, 2019-01-02:
When you use esptool, pull GPIO2 to GND before pushing reset. Otherwise the ESP32 will not be in bootloader mode. Some breakout board have circuitry included which does that for you.

misan128, 2019-01-02:
> You've bricked it. I've also managed to do this couple times in past.
> I've managed to erase flash with esptool, but you need to be fast:
> 
> 1. Connect your nodeMCU
> 2. prepare esptool cmd in terminal/cmd
> 3. Press reset button on nodeMCU and exec (2) almost the same time.
> 
> If failed - repeat (3).

I'm trying this but it's not working. Theres any kind of trick to do it?

misan128, 2019-01-02:
> When you use esptool, pull GPIO2 to GND before pushing reset. Otherwise the ESP32 will not be in bootloader mode. Some breakout board have circuitry included which does that for you.

I'm doing this but it starts transmiting uncontrollably!

dpgeorge, 2019-06-24:
This sounds like an issue flashing the board which is not related to MicroPython, so closing the issue.

BTW, there is now a simpler way to deepsleep for a period of time: `machine.deepsleep(10000)`

closed 2019-06-24 by dpgeorge