micropython/micropython #4391: Questions SMTM32F207 support

duaneellissd, 2019-01-02:
Been playing and working towards support for the STM32F207 Nucleo board
I have some questions in this regard

Question (1) - Do you have a checklist for things to do when making a new board? And new chip?

Question (2) What is done to test today In some automated way with actual hardware? 

There are different ways of looking at this problem:

(A) Language issues - many things can be found using/forcing x86 (32bit) linux.
     (which is not hard, just compile with GCC using "-m32")

(B) ARM cortex (or other cpu, ie: ESP8266, or RiskV, etc) specific
      A barebones simulation running the language tests would work well.

(C) Target/board specific - which is my current focus

I've done things where I have two boards: (A) can be the stimulus, and (B) can be the DUT
If both run uPython - that makes it very easy to write tests on both sides.

More so - if there is a way to download small test scripts to the DUT and the stimulus device.

Any ideas/suggestions/comments?

-Duane.

chrismas9, 2019-01-09:
`tests/pyb` is a good start but i have found it necessary to do hardware tests with external connections to find all issues. For example I am doing an F413 port and UARTs 9 & 10 worked but at the wrong baud rate. This was because they are on APB2 but the STM32 HAL hadn't put them on CLK2. This failure can only be detected with an external connection to an FTDI cable or oscilloscope to check the baud rate.

i also found issues with the interrupt vector table. This doesn't cause compile errors, only run time errors with hardware test scripts.

dpgeorge, 2019-01-17:
Hopefully the above helps.  Further questions should be directed to https://forum.micropython.org, or more specific issues opened here.

closed 2019-01-17 by dpgeorge