micropython/micropython #4394: esp32 localtime and machine.RTC().datetime() inconformity

junhuanchen, 2019-01-03:
see this 

https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/ports/stm32/modutime.c#L84

and this

https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/ports/esp32/machine_rtc.c#L105

I know it seems to have been designed, but if I assign values like this, I won't be able to use the following code.

```python
machine.RTC().datetime(time.localtime(546450051)) # 546450051 only a number.
```

Is there a better way to set seconds to RTC time.

Otherwise I'll fill in the third element to align the structures, is that so?

junhuanchen, 2019-01-03:
[image]

robert-hh, 2025-10-17:
Most RTC devices using in the various ports do not support subseconds. And the integer times handled by gmtime() and mktime() have seconds resolution. So fractional seconds precision is hardly achieved. But if you want the RTC time be in sync with the time received by NTP, why not straight set it to that NTP time, like it is done in ntptime.py? The skips at least steps 1 - 3.

jamesquilty, 2025-10-17:
@robert-hh Thanks for responding, your input is always appreciated!

I think that the time tuple inconsistency is a bug that needs to be fixed. The example I gave is only an example, others have reported other examples where the inconsistency needs to be worked-around. I'm happy to discuss my application (below), but the specifics might be a little bit of a distraction from what I'd consider the core bug: there's a time tuple inconsistency.

My application uses an ESP32 which does support fractional seconds - and it's very useful because it's advantageous for my application for the RTC to be accurate to ±0.5 seconds of UTC or better. To work with fractional second precision it's necessary to use fixed-point representation on the ESP32 because floating point doesn't accurately represent fractional times when the integer part is on the order of $`8\times10^8`$ for the ESP32 epoch and $`3\times10^10`$ for NTP timestamps. That `time.mktime()` only returns seconds isn't a problem at all. I've been able to achieve accuracy to within ±0.01 s using SNTP. I don't use the approach in [ntptime.py](https://github.com/micropython/micropython-lib/blob/master/micropython/net/ntptime/ntptime.py) because it doesn't perform full SNTP offset calculation ([RFC 4330 Section 5](https://www.rfc-editor.org/rfc/rfc4330#page-14)) and won't yield the desired accuracy. That the ESP32 RTC provides fractional seconds certainly makes it easier to implement SNTP client calculations. 

FWIW, I have battery-backed external (to the ESP32) RTC (DS1307 or DS3231) which don't expose the fractional second count via their registers but which can easily be polled for roll-over to determine offsets and which reset their fractional second counter when the seconds register is written, which makes it pretty easy to synchronise the external RTC to the ESP32 without losing much (if any) accuracy to UTC. My application has only a cellular IoT modem (no WiFi connectivity) which has an internal RTC which obtains time via NITZ after operator registration, but the modem RTC doesn't support fractional seconds, the NITZ-provided accuracy to UTC is ±1 second _at best_ and the modem RTC drifts between registrations in a way that's difficult to measure via AT commands. All of this has been the motivation to obtain time via SNTP.