micropython/micropython #4397: ujson.dumps() not handling non-ascii characters correctly

martijnthe, 2019-01-04:
`ujson.dumps` does not appear to properly escape non-ascii characters and generated JSON that is not up to the spec (and fails to parse in a web browser context for example).

```
$ ./micropython
MicroPython v1.9.4-771-gb33f108cd-dirty on 2019-01-04; darwin version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> from ujson import dumps
>>> dumps("\U0001F602")
'"\U0001f602"'
>>> dumps("\u2082")
'"\u2082"'
>>> print("\u2082")
₂
>>> print("\U0001F602")
😂
```

See source code: https://github.com/micropython/micropython/blob/5a91fce9f868eeba37a0b1cf6c3b4435ed5eecec/py/objstr.c#L95

For the emoji example (outside the BMP), the code point should be escaped as a surrogate pair `"\uD83D\uDE02"`.
The subscript two should be escaped as `"\u2082"`.

I'm happy to fix it, but some guidance would be nice around these questions:
1. IMHO, JSON unicode escaping should happen, even if `MICROPY_PY_BUILTINS_STR_UNICODE` is not defined. Agree?
2. There's already some utf-8 decoding code in the loop in `uni_print_quoted`, but it's not purely utf-8 decoding that is happening there (it's mixed with somewhat python specific backslash-escaping stuff). So I'm not sure if it's worth extracting that. Since it's just a few lines of duplication, I'm thinking of basically copy the relevant parts over, plus adding the additional handling for codepoints outside of the BMP and emit surrogate pairs.
3. Why is there no test expectation for `ujson_dumps.py`? Can I add one?

martijnthe, 2019-01-04:
@pfalcon thanks for the quick reply.

My bad, I misinterpreted the [spec] / missed crucial parts... TIL, thanks!

[spec]: https://www.ecma-international.org/publications/files/ECMA-ST/ECMA-404.pdf

closed 2019-01-04 by martijnthe

pfalcon, 2019-01-04:
Well, there's definitely differences in MicroPython vs CPython output for non-ascii chars:

~~~
MicroPython v1.9.4-908-gf4f3e2b60-dirty on 2019-01-04; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ujson
>>> ujson.dumps("\u2082")
'"\u2082"'
~~~

vs

~~~
Python 3.6.7 (default, Oct 22 2018, 11:32:17) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import json
>>> json.dumps("\u2082")
'"\\u2082"'
~~~

But from quick at Wikipedia (not a normative spec, yeah):

> JSON exchange in an open ecosystem must be encoded in UTF-8.[18] The encoding supports the full Unicode character set, including those characters outside the Basic Multilingual Plane (U+10000 to U+10FFFF). However, if escaped ...

So, MicroPython chooses not to escape unicode, and then everything should be indeed ok.