micropython/micropython #4399: Some processes lock at timer single shot executed after machine.reset()

nedoskiv, 2019-01-06:
Hello, it is kind a complicated to explain how this bug works, but I will try :)

i'm using micropython 1.9.4 on esp32 (lolin32 lite)

in my startup script I got script that initialize wireless connection and shot single timer to show connection status after 10 seconds:
```
def ws(timer):
	if station.isconnected() == False:
		print ("Wifi still not connected")
	else:
		print(station.ifconfig())
	timer.deinit()
	exec(open('/alarmprocessor.py').read(),globals())
timer.init(period=10000, mode=machine.Timer.ONE_SHOT, callback=ws)

```
after that I got few scripts that go into different threads and main is in a loop and display some data in console.
So far so good, all work, but when I interrupt main script with CTRL-C and execute machine.reset()
then start my startup script again it loads, main script start but in the moment when timer should display his message about wireless connection, main script hands and some of _thread scrips also stop working.
If I restart device using reset button it is ok and my start scripts load normally (just first time)
Now I was removed that timer.single.shot and all works fine after machine.reset()

Hope  managed to explain bug that I have found.

dpgeorge, 2019-01-09:
How exactly are you creating the `timer` object?  It seems that `machine.reset()` does not reset legacy timers, so if you use one of those that may be the problem.

nedoskiv, 2019-01-11:
timer = machine.Timer(0)

dpgeorge, 2019-01-16:
I can't reproduce this issue.  Using the following code:
```python
timer=machine.Timer(0)
timer.init(period=1000, mode=machine.Timer.ONE_SHOT, callback=lambda x:print(x))
```
The callback is called correctly, then I do `machine.reset()`, then run the above code again, and the callback is called correctly again.

----

@nedoskiv I suspect there could be an interaction here with timer and threads.  Can you please try your code with the latest esp32 firmware (available at https://micropython.org/download#esp32) and see if it still has the error?  If so, please post minimal code that reproduces the problem, thanks!

nedoskiv, 2019-01-17:
Very strange, I tried my old code again and it locks, difference is that i have few threads running, gonna try with/without them and post the results

nedoskiv, 2019-01-17:
Ok I spend few hours in attempt to isolate how/when that happen and fail, when all of my components are loaded (spi, interface, tcp usocket server, and few other things) it fails as I described. So I start to cut off one by one and try to find exactly where it start to happen, when go to GPIO/ADC initialization it stop to fail, after that I add that part again and it continue to work fine, so I give up after 3h :), gonna close that issue and reopen it when find right place when that start to happen, or if you wish I can give you complete source code in an archive and then you gonna reproduce the problem but  as I said, hard to isolate what cause it.

Thank you for your time.

closed 2019-01-17 by nedoskiv

dpgeorge, 2019-01-28:
I found an issue closely related to this one.  Making a periodic timer and doing a soft reset could lead to a crash.  Eg:
```python
machine.Timer(-1).init(period=1000, callback=print)
```
That will print the timer object once per second.  Then doing a ctrl-D to soft reset the board would lead to a crash (or invalid TypeError).  That issue is now fixed by 9ac9aa989c89a1d6310160dd408820d1164d4dd7, which may fix the original issue reported here.