micropython/micropython #4403: MicroPython website DNS/hosting issue?

mcauser, 2019-01-09:
Getting a `ERR_CONNECTION_REFUSED` error on:
http://micropython.org/
http://forum.micropython.org/
http://docs.micropython.org/
http://georgerobotics.com/

https://downforeveryoneorjustme.com/micropython.org
> Is micropython.org down?
> It's not just you! micropython.org looks down from here.

DNS A records pointing to 176.58.119.26, which is responding to pings.

mcauser, 2019-01-09:
Services back to normal.

dpgeorge, 2019-01-09:
Yes, there was a brief downtime for maintenance.

dpgeorge, 2019-01-13:
All sites are currently down for further maintenance.

dpgeorge, 2019-01-13:
Maintenance has finished for now and most services should be back to normal, but please report any problems here.

dpgeorge, 2019-01-13:
The wiki.micropython.org is known to be still down.

dpgeorge, 2019-01-14:
wiki.micropython.org is back online.

flowergrass, 2019-01-20:
http://micropython.org/unicorn/ is broken D:

dpgeorge, 2019-01-21:
> http://micropython.org/unicorn/ is broken D:

Thanks for hint!  Serving of static .gz files was disabled with the new server.  It's now enabled and should be working again.

mattytrentini, 2019-01-21:
All service seem to be up; can this ticket be closed now?

dpgeorge, 2019-01-22:
There are still a few things not running, like doc rebuilding.

dpgeorge, 2019-01-22:
Docs and firmware builds are now working again, and they should be the remaining items.

closed 2019-01-22 by dpgeorge