micropython/micropython #4406: socket OSError: 128 after unreachable connection attempt

siochs, 2019-01-10:
Hi,
using ESP32 on an Heltec WiFi 32 board. Micropython nightly build from yesterday.
I am setting up wifi to turn on the network interface:
```
>>> import network
>>> ap = network.WLAN(network.AP_IF)
>>> ap.active(True)
>>> ap.config(essid='ssid', authmode=network.AUTH_WPA_WPA2_PSK, password='password') 
>>> ap.ifconfig(('192.168.84.1', '255.255.255.0', '192.168.84.1', '192.168.84.1')) 
```
This turns on the access point and i can connect to it. I can see it working from the output.
Now try this:
```
>>> import socket
>>> s = socket.socket()
>>> s.connect(("192.168.84.99", 80)) # there is no client having this ip and no service at port 80
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 113] EHOSTUNREACH # this was expected
>>> # RETRY
>>> s.connect(("192.168.84.99", 80)) # does not matter if you use a working ip/service here!
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 128
```
So after a connection error, the socket instance is completely unusuable. I have to make a new socket.socket() to start over. Since sockets are system resources and costly on creation, i think this is not an acceptable workaround.
However this is the CPython behaviour:
```
>>> import socket
>>> s = socket.socket()
>>> s.connect(("192.168.84.99", 80))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TimeoutError: [WinError 10060] Unreachable
>>> # RETRY
>>> s.connect(("192.168.84.99", 80))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TimeoutError: [WinError 10060] Unreachable
```
So it seems sockets should be reusable after a connection errror.

pfalcon, 2019-01-11:
> In the docs I read that it should be avoided to destroy and recreate instances as much as possible (heap corruption).

Which docs did you read that ("heap corruption")? And did you consider to stop worrying and just write your software and worry about the actual problems you face (if any)?

Or if you're interested in a total and generic solution to this problem, there's an RFC how to do that: https://github.com/micropython/micropython/issues/4331

siochs, 2019-01-11:
@pfalcon Sorry, wrong term. I meant heap fragmentation.
[Micropython on microcontrollers / RAM](https://docs.micropython.org/en/latest/reference/constrained.html#ram) says the following:
> In addition to memory consumption, there is also an issue known as heap fragmentation. In general terms it is best to minimise the repeated creation and destruction of objects. The reason for this is covered in the section covering the heap.

By the way, I never just go ahead and write my software I always start writing tests to see if my requirements (concerning the platform and concerning the code) can be met. And now I ran into THIS problem forcing me to permanently recreate socket instances. After what I read in the RAM section and because acquiring system resources appeared expensive I started "worrying", yes.

pfalcon, 2019-01-13:
> And now I ran into THIS problem forcing me to permanently recreate socket instances.

SOCK_STREAM type sockets generally designed in a way that a new socket is required for *each* connection. And a connection generally can be ended by reasons you don't control at any time. So, as long as you work with connection-oriented sockets, you should be prepared to recreate sockets all the time. Again, to address cases like that, https://github.com/micropython/micropython/issues/4331 is being proposed (as a non-standard MicroPython extension).

siochs, 2019-01-14:
@pfalcon I see. Good purpose by the way (#4331). Reuse of "closed" object would also be great for file, 
```
f = open("file.txt")
# do something
f.close()
# so something
f.reopen() #<- would be nice
```

dpgeorge, 2019-01-16:
> so that would mean when changing the framework to lets say C++/Arduino the same problem would occur? You say, fixing the ESP IDF is possible, is there a chance to open an issue somewhere?

If it's possible to reproduce this issue at the C/C++ level via a program written using just the ESP IDF, then it's very likely it could be fixed at that level through the repo https://github.com/espressif/esp-idf/

closed 2019-01-16 by siochs