micropython/micropython #4408: Files are missing boot.py and main.py on ESP8266 board

lazarvgd, 2019-01-13:
Hi there, I am using esp8266 lolin with micropython. I have done some kind of stress tests, disconnecting the power, resetting the board with reset button in order to test how it will behave.

What have I found is that files are missing from the board. I am using **ampy** to check the files on board. I have executed the commnad `sudo ampy --port /dev/tty.wchusbserial1420 -d1 ls` and I have gotten the output like this:
`/.
/.
/.
/.
/.
/.
/.
/.
/.
/.`

I am not sure why is this happening, I was able to upload files, get existing files without any problem until I once pressed reset button to see will the board continue with work as expected.

Can somebody tell me is this related to micropython or the chip quality?

Thanks :)

robert-hh, 2019-01-13:
You might have corrupted the file system with your stress test. The device uses a FAT file system, which is not robust to power failures. As a cure erase the flash and re-load firmware & files.

I would not call that an issue, but more a property. A different file system would definitely be better, and that may see the light of day sometime.

lazarvgd, 2019-01-13:
> You might have corrupted the file system with your stress test. The device uses a FAT file system, which is not robust to power failures. As a cure erase the flash and re-load firmware & files.
>
> I [...]

Thanks, I did not know that.

closed 2019-01-13 by lazarvgd

mcauser, 2019-01-13:
A flakey old USB cable has corrupted a few of my file systems like this. Fixed with an erase flash then write flash. Then copy the .py files back.

lazarvgd, 2019-01-14:
> A flakey old USB cable has corrupted a few of my file systems like this. Fixed with an erase flash then write flash. Then copy the .py files back.

Well I have problem with power supply, any idea how to make esp more resilient to these problems?

thanks

robert-hh, 2019-01-14:
Close open files as quick as possible. 
Fix the power supply. 
Set up your power supply to give an indication of power failure early enough (a few ms), in order to interrupt the ESP for a controlled shutdown.

lazarvgd, 2019-01-14:
> Close open files as quick as possible.
> Fix the power supply.
> Set up your power supply to give an indication of power failure early enough (a few ms), in order to interrupt the ESP for a controlled shutdown.

Thank you very much.