micropython/micropython #4409: Is the website down?

ravernkoh, 2019-01-13:
Hi.

Is the website currently undergoing maintenance or is it down? Trying to access https://micropython.org/ results in an error where the TLS certificate is rejected. Proceeding anyway will result in the following screen:

[image]

I need to download the ESP32 firmware urgently.

robert-hh, 2019-01-13:
I prfepared two images  firmware_esp32_no_spiram.bin and firmware_esp32_with_spiram.bin here:
https://github.com/robert-hh/Shared-Stuff

dpgeorge, 2019-01-13:
Please see #4403.  It should be back up now.

closed 2019-01-13 by dpgeorge