micropython/micropython #4410: Pin.off no attribute 'off' / same with .on

stefsolar, 2019-01-13:
Dear MicroPython-team,
first of all, thanks for this super great project.

I was just playing with my ESP and realized I could not use the Pins as I would expect to?
My example is something like:
`from machine import SPI, Pin
spi = SPI(1, baudrate=1000000, sck=18, miso=19, cs=5, mosi=23)
cs = Pin(5, Pin.OUT)
cs.off()

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Pin' object has no attribute 'off'

cs.on()

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Pin' object has no attribute 'on'
`

Any idea on this? I feel I am missing something?
Best, stefey

robert-hh, 2019-01-13:
For unknown reasons the on/off methods do not exist in the ESP32 port. You may use the pin.value(1) or pin.value(0) or just pin(1) and pin(0). Personally I did never understand the advantage of on/off, because whether something like a LED is on of off depends on the way it is connected.

dpgeorge, 2019-01-16:
Thanks for the report, this is definitely an omission.  Fixed by 90f86a0197f8acb53add617f48a43947c2d5e223

closed 2019-01-16 by dpgeorge

stefsolar, 2019-01-21:
Thanks a lot guys :-)