micropython/micropython #4411: f presentation type is not rounding

qoheniac, 2019-01-19:
In MicroPython (tested with version 1.9.3) the f presentation type (as in `"{:.0f}".format(x)`) is not rounding floating point numbers as CPython (tested with 3.7.2) does. Actually I'm not sure what it really does. It is not rounding, not flooring and not truncating as the following examples show:

    print("{:.0f}".format(100.0))
    CPython: >>> 100
    MicroPython: >>> 100

    print("{:.0f}".format(100.9))
    CPython: >>> 101
    MicroPython: >>> 100

    print("{:.0f}".format(101.0))
    CPython: >>> 101
    MicroPython: >>> 100

    print("{:.0f}".format(99.5))
    CPython: >>> 100
    MicroPython: >>> 99

    print("{:.0f}".format(99.6))
    CPython: >>> 100
    MicroPython: >>> 100

dhylands, 2019-01-20:
I get slightly different results from yours. I tested with:
```
MicroPython v1.9.3-521-gd6cf5c674 on 2018-05-22; PYBv1.1 with STM32F405RG
```
```
>>> '{:.0f}'.format(100.9)
'101'                     
>>> '{:.0f}'.format(100.0)
'100'                     
>>> '{:.0f}'.format(100.9)
'101'                     
>>> '{:.0f}'.format(101.0)
'101'                     
>>> '{:.0f}'.format(99.5) 
'99'                      
>>> '{:.0f}'.format(99.6) 
'100'                     
```
The difference with the 99.5 case is most likely due to the fact that 99.5 doesn't have an exact representation. MicroPython is using 32-bit floats, where CPython is using 64-bit floats. For example:
```
>>> '{:.8f}'.format(99.5)
'99.49999809'
```

qoheniac, 2019-01-20:
I also used a pyboard version 1.1 with a STM32F405RG for testing and thought of something like 32-bit versus 64-bit floats at first too. But that doesn't explain why mine printed out 100 for 101.0 and 100.9, right? Interesting that it doesn't happen with your board.

qoheniac, 2019-01-20:
I think I got the pattern: all numbers between -100 and 100 are rounded just like CPython does while the rest of the numbers get truncated i.e. numbers above 100 get rounded down and numbers below -100 get rounded up. But now that I see the pattern it still doesn't make any sense to me. The thing with 101 becoming 100 is because of the floating number representation (101 -> 100.99... -> 100).
[image]

jonnor, 2024-09-19:
Is this issue still relevant on recent MicroPython versions?

That single precision floating point is used is at least expected behavior.