micropython/micropython #4412: ustruct packing alignment issues

deets, 2019-01-20:
I've run into an issue today where unpack_into failed me.

The ESP32 apparently has some memory alignment issues that I stumbled upon in some C-written projects. I now encountered them using the ustruct module. The problem appears to be that e.g. packing a uint32_t into a memory-location not aligned to 4 bytes will force the write to the *next* memory address properly aligned.

See this example:

```
>>> import ustruct
>>> buffer = bytearray(24)
>>> ustruct.pack_into('I', buffer, 0, 1234)
>>> buffer
bytearray(b'\xd2\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
>>> ustruct.pack_into('I', buffer, 2, 1234)
>>> buffer
bytearray(b'\xd2\x04\x00\x00\xd2\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
```
As one can clearly see, the bytes 0xd2 and 0x04 were expected at offset 2, but start in fact at offset 4.

My version of micropython is a customised one but based off v1.9.4.

pfalcon, 2019-01-20:
See https://docs.python.org/3/library/struct.html#byte-order-size-and-alignment , especially pay attention to "Alignment: native" table cell.

deets, 2019-01-20:
Interesting. Thank you, I close this as a non-issue.

closed 2019-01-20 by deets

deets, 2019-01-20:
Yep, I figured that one out. After I got over missing = ;)