micropython/micropython #4414: Propose integration of pulseio functions

ruky87, 2019-01-21:
It should be useful to integrate the "pulse" functions of Circuitpython and the infrared driver, with these functions it's possible to read and send IR codes with a GPIO pin for example. These functions permits to send a sequence of pulse and read a sequence of pulses (and relative duration). It is possible to generate tones or read some types of sensors.
I think that there is also other stuff from Circuipython that could be integrated in the main Circuitpython (at least the architecture independent stuff)

pfalcon, 2019-01-23:
MicroPython already offers a function to time a pulse: http://docs.micropython.org/en/latest/library/machine.html?highlight=machine#machine.time_pulse_us . Generating pulses is as simple as:

~~~
pin.on()
utime.sleep_us(10)
pin.off()
~~~

> I think that there is also other stuff from Circuipython that could be integrated 

Absolutely, there's other stuff in CircuitPython and in many other places. But there's no need to "integrate" all that stuff into **Micro**Python. Let it live in separate modules, there's a framework for that already: https://github.com/micropython/micropython/pull/4195

ruky87, 2019-01-23:
For generating pulses this is a way, but the precision is lower respect to the Circuitpython implementation.
But the main problem is the pulse recording (for example when reading an IR or RF remote code): you have to record pulse length for each pulse with high precision, interrupts need to be managed in C because of the reduced latency (https://github.com/adafruit/circuitpython/blob/master/ports/esp8266/common-hal/pulseio/PulseIn.c). I think that with these modules it's easier to interface with some type of sensors, generate tones, etc. It's like the OLED driver integrated in Micropython, it is there because it's useful I think. Micropython has all the basic features of Arduino, if I go on the Arduino reference under language functions section I can see that more or less all the stuff can be done with Micropython except generating and recording pulse trains.
https://www.arduino.cc/reference/en/language/functions/advanced-io/pulsein/
Micropython is used mainly for IoT, robotics... So interfacing easily with sensors is a good thing. There's no need to integrate all that stuff, but some extra stuff could be useful, especially for people used with arduino-like functions.

peterhinch, 2019-01-24:
Decoding IR with MicroPython is straightforward. See [this solution](https://github.com/peterhinch/micropython-async/tree/master/nec_ir) which uses a pin interrupt, **utime** and **uasyncio**.

ruky87, 2019-01-24:
Are there still problems with the interrupt latency with Micropython on ESP boards?

gpshead, 2019-01-27:
FWIW, I have to agree agree that the `pulseio` C module is better than writing an interrupt handler in micropython itself (or using a library that does).  Both for timing reasons and the fundamental problem that ISRs in Python have way too many crash-worthy constraints (["avoid memory allocation"](https://docs.micropython.org/en/latest/reference/isr_rules.html#tips-and-recommended-practices)... um, okay, but **this is a high level language** - the programmer does not control the VM that way and isn't supposed to know when such things _might_ happen).  Interrupt handlers are not friendly. They also don't even appear to have a standard interface. The arecord.py code has three branches based on hardware platform to setup an irq... ugh.

I've got a sensor that requires pulse width occupancy measurements with the smallest pulse being 10us. I'd never trust a Python ISR for that kind of timing. Especially on devices time sharing the cpu with soft device radio firmware where you need a quick priority interrupt to record the event that won't interfere with radio timing.