micropython/micropython #4417: LittlevGL and μGFX support for uPython on ESP32.

beyonlo, 2019-01-21:
Hi all!

1. I would like to discuss with you about that great GUI libs for ESP32, what you think about?
2. Is a good idea to support it on uPython on ESP32?
3. Is need a hard work to support it? Please, correct me if I'm wrong, but if that lib is pure C, to do a interface to uPython is not so hard, right? How that works?

https://www.espressif.com/en/news/ESP32_Modules_Now_Support_LittlevGL_and_%CE%BCGFX

Thank you.

amirgon, 2019-01-26:
There is an effort being done on LittlevGL side to provide Micropython bindings (module).  

More information:

- [Initial version of README ](https://github.com/littlevgl/lv_bindings/blob/cff4147a992422207720bd8cb02e78f34070a228/README.md)
- [Open issue](https://github.com/littlevgl/lvgl/issues/557)
- [Micropython Bindings repo](https://github.com/littlevgl/lv_bindings)
- [Micropython fork with lvgl bindings](https://github.com/littlevgl/lv_mpy). Currently only unix port is running lvgl but I plan to add support for ESP32 too.

jonnor, 2023-07-30:
LVGL support for ESP32 with MicroPython is pretty mature now, and is linked in the post above.
Proposing to close this issue.

closed 2023-08-03 by jimmo