micropython/micropython #4422: chr() leaks

pmp-p, 2019-01-22:
```
#
# to run with small heap eg -X heapsize=16384
#

try:
    import utime as time
except:
    import time

import gc
cnt = 1
last = 0

while cnt < 0x110000:
    time.sleep(.005)
    current = gc.mem_free()
    #GOOD print( '[%c]' % cnt , current - last, current )

    #BAD print( '{}'.format( chr(cnt) ), current - last, current )

    #BAD
    print( chr( cnt ), cnt , current - last, current )
    cnt +=1
    last = current
    gc.collect() #no effect maybe a useless volatile interned strings of length 1 somewhere ?

```

pfalcon, 2019-01-22:
~~~
"".join([chr(x//3) for x in (261, 312, 291, 348, 117, 345, 96, 348, 312, 315, 345, 189)])
~~~

pmp-p, 2019-01-22:
Title is explicit enough no ?
run code, enjoy MemoryError after a while on any port, unix port is good enough for that  or pycopy with a esp sized heap.

pfalcon, 2019-01-22:
Look how much info can be packed into human words ;-).

> or pycopy with a esp sized heap.

Good mentioning, because aggressive string interning introduced by @dpgeorge was one of the few patches I reverted (so far) in my fork.

But otherwise, yeah, looking at mp_builtin_chr(), it interns unconditionally. A solution would be interning only ASCII stuff by default, allow to disable interning in optional cases (like this) altogether.

pmp-p, 2019-01-22:
"unconditionally" => that may explain why my doubts probably left me speechless at first.

- short strings should not be interned at all.
- at worst, interning should be kept on MRU cache decisions not unconditionally.
- at best interning really long strings should be paged to flash ( like arduino progmem ? ) on user request, maybe via a custom type or simply just with f"" ( like cpython f-strings syntax )  ?.

pfalcon, 2019-01-22:
> short strings should not be interned at all.

Obviously the opposite. And quite obviously actually:

* Shorter strings have a higher chance of reuse (and that's why they're interned at all).
* Allocating short strings has highest overhead. E.g. for 1 useful char, a whole 16-byte alloc block is used, 93% wasted. Surely, not allocating them like that is better.

pmp-p, 2019-01-22:
by short i meant len() < sizeof(  hash + storage pointer )

pmp-p, 2019-01-22:
thx for incrementing the reference counter ;)

pmp-p, 2019-01-24:
a tested working fix is available here https://github.com/pfalcon/pycopy/commit/4d7a02654d37024d73f83d47f992f2dd91e2d8b7.diff

jonnor, 2024-09-19:
No MemoryError when running this on Unix port with version v1.24.0-preview.276.g1897fe6227. But it seems to stop at 

```
154 0 11296

 156 0 11296
```
which seems way to early?