micropython/micropython #4423: Cannot send a variable on UART

hetvishah08, 2019-01-22:
I am trying to send a value stored in a variable through UART but it shows some error messages.I tried to send a list on UART as well and it showed me the same error.

**Code and Debug using REPL mode:**

>>> from machine import UART
>>> u2=UART(2,9600)
>>> u2.init(9600,bits=8,parity=None,stop=1,tx=10,rx=9)
>>> var=123
>>> u2.write(var)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: object with buffer protocol required
>>> 
>>> 
 How do I send a variable on UART?

peterhinch, 2019-01-22:
This should be raised in [the forum](http://forum.micropython.org/) not the bug tracker. Clue: the UART write method expects a **bytes** or **bytearray** instance, not an integer.

hetvishah08, 2019-01-22:
Thank you i was able to send data using bytearray() but what for the float type data? bytearray() doesn't have float type so can i send variable of data type float?

robert-hh, 2019-01-22:
You have to convert other data to a bytearray, like with:
`str(data).encode()`

closed 2019-01-23 by dpgeorge