micropython/micropython #4425: esp8266 recvfrom() gives ap address instead of source

pmp-p, 2019-01-24:
on esp8266 with master :
STA 192.168.4.2 send packet to AP with sendto( data, ('192.168.4.1',port) )

on ap data, adrr = recvfrom() gives addr[0] == '192.168.4.1' source address instead of '192.168.4.2'

dpgeorge, 2019-01-25:
I can't reproduce this.  In my test, recvfrom() on the AP returns 192.168.4.3 for the IP of the connected STA.

To make progress here, please provide full working code for both sides of the connection.

dpgeorge, 2019-02-21:
I assume this was resolved.  Feel free to reopen with further info how to reproduce the problem.

pmp-p, 2019-05-10:
sorry i don't have time to write test code yet and i usually close my bugs when they are solved,  but i highly suspect it has to do with switching from STA to AP at runtime.

dpgeorge, 2019-05-13:
> i highly suspect it has to do with switching from STA to AP at runtime.

If you can expand on this and provide some pseudo code for a possible test for this case then that would be helpful.