micropython/micropython #4429: lwip git submodule URL outdated and non-functioning

SpotlightKid, 2019-01-25:
The git repo URL of lwip is now https://git.savannah.nongnu.org/git/lwip.git

See http://savannah.nongnu.org/git/?group=lwip

Currently one is unable to build MicroPython from a repo clone, because the lwip submodule can not be checked out.

Unless you do:

    sed -i -e 's|gnu\.org/r|nongnu.org/git|' .gitmodules
    git submodule update --init

after cloning the repo.

dpgeorge, 2019-01-25:
> Currently one is unable to build MicroPython from a repo clone, because the lwip submodule can not be checked out.

I'm able to checkout this lwip submodule without any issue (I just tried with a fresh clone).

SpotlightKid, 2019-01-25:
Hmm, there seems to be a problem with the resolver of my local git.

```$ git clone https://github.com/micropython/micropython.git
Klone nach 'micropython' ...
remote: Enumerating objects: 87, done.
remote: Counting objects: 100% (87/87), done.
remote: Compressing objects: 100% (38/38), done.
remote: Total 67713 (delta 55), reused 65 (delta 49), pack-reused 67626
Empfange Objekte: 100% (67713/67713), 38.05 MiB | 2.22 MiB/s, Fertig.
Löse Unterschiede auf: 100% (48761/48761), Fertig.
$ cd micropython/
$ LC_ALL=C git submodule update --init
Submodule 'lib/axtls' (https://github.com/pfalcon/axtls) registered for path 'lib/axtls'
Submodule 'lib/berkeley-db-1.xx' (https://github.com/pfalcon/berkeley-db-1.xx) registered for path 'lib/berkeley-db-1.xx'
Submodule 'lib/libffi' (https://github.com/atgreen/libffi) registered for path 'lib/libffi'
Submodule 'lib/lwip' (https://git.savannah.gnu.org/r/lwip.git) registered for path 'lib/lwip'
Submodule 'lib/nrfx' (https://github.com/NordicSemiconductor/nrfx.git) registered for path 'lib/nrfx'
Submodule 'lib/stm32lib' (https://github.com/micropython/stm32lib) registered for path 'lib/stm32lib'
Cloning into '/home/chris/tmp/micropython/lib/axtls'...
Cloning into '/home/chris/tmp/micropython/lib/lwip'...
fatal: unable to access 'https://git.savannah.gnu.org/r/lwip.git/': Could not resolve host: git.savannah.gnu.org
fatal: clone of 'https://git.savannah.gnu.org/r/lwip.git' into submodule path '/home/chris/tmp/micropython/lib/lwip' failed
Failed to clone 'lib/lwip'. Retry scheduled
Cloning into '/home/chris/tmp/micropython/lib/berkeley-db-1.xx'...
Cloning into '/home/chris/tmp/micropython/lib/stm32lib'...
Cloning into '/home/chris/tmp/micropython/lib/lwip'...
fatal: unable to access 'https://git.savannah.gnu.org/r/lwip.git/': Could not resolve host: git.savannah.gnu.org
fatal: clone of 'https://git.savannah.gnu.org/r/lwip.git' into submodule path '/home/chris/tmp/micropython/lib/lwip' failed
Failed to clone 'lib/lwip' a second time, aborting
Cloning into '/home/chris/tmp/micropython/lib/nrfx'...
Cloning into '/home/chris/tmp/micropython/lib/libffi'...
```

But when I do a lookup on `git.savannah.gnu.org` everything is fine:

``$ nslookup git.savannah.gnu.org
Server:		192.168.42.1
Address:	192.168.42.1#53

Non-authoritative answer:
Name:	git.savannah.gnu.org
Address: 209.51.188.201
Name:	git.savannah.gnu.org
Address: 2001:470:142:5::201
```

I'll have to do some digging, why this is.

Anyway, the repo URL still is outdated, isn't it?

SpotlightKid, 2019-01-25:
After testing this for a while, suddenly I can't reproduce this anymore. Seems to have been a temporary problem with one of my upstream nameservers.

Leaving this open, in case you want to update the lwip repo URL to `nongnu.org`. Feel free to close otherwise.

pfalcon, 2019-01-25:
There're regular intermittent service interruptions with savanna site. Please use the build history at https://travis-ci.org/micropython/micropython as an indicator whether something works or not.

dpgeorge, 2019-01-27:
@SpotlightKid do you know if there was an announcement or something about the Savannah URLs being changed?  Or was this URL outdated for a long time?

mrohit2011, 2019-01-29:
I too got the same issue.

$ git submodule update --init
Cloning into 'C:/Users/Rohit/Desktop/micropython/micropython/lib/lwip'...
fatal: unable to access 'https://git.savannah.gnu.org/r/lwip.git/': SSL certificate problem: self signed certificate in certificate chain
fatal: clone of 'https://git.savannah.gnu.org/r/lwip.git' into submodule path 'C:/Users/Rohit/Desktop/micropython/micropython/lib/lwip' failed
Failed to clone 'lib/lwip'. Retry scheduled
Cloning into 'C:/Users/Rohit/Desktop/micropython/micropython/lib/lwip'...
fatal: unable to access 'https://git.savannah.gnu.org/r/lwip.git/': SSL certificate problem: self signed certificate in certificate chain
fatal: clone of 'https://git.savannah.gnu.org/r/lwip.git' into submodule path 'C:/Users/Rohit/Desktop/micropython/micropython/lib/lwip' failed
Failed to clone 'lib/lwip' a second time, aborting

dpgeorge, 2019-01-29:
As far as I can tell both `git.savannah.gnu.org` and `git.savannah.nongnu.org` point to the same server (same IP address), and whether the URL uses `/r/` or `/git/` just means whether it's the read-only version or not.  So the following all give equivalent repos:
```
git clone https://git.savannah.gnu.org/r/lwip.git
git clone https://git.savannah.gnu.org/git/lwip.git
git clone https://git.savannah.nongnu.org/r/lwip.git
git clone https://git.savannah.nongnu.org/git/lwip.git
```
And they all work ok for me right now.  The uPy submodule references the first in this list.

mrohit2011, 2019-01-31:
> I too got the same issue.
>
> $ git submodule update --init
> Cloning into 'C:/Users/Rohit/Desktop/micropython/micropython/lib/lwip'...
> fatal: unable to access 'https://git.savannah.gnu.org/r/lwip.git/': [...]

'https://git.savannah.gnu.org/r/lwip.git' was blocked on my network, hence I wasn't able to update lwip module. Now since I got it unblocked, it's working fine.

dpgeorge, 2019-02-21:
Closing because the lwip URL is still valid.

closed 2019-02-21 by dpgeorge