micropython/micropython #4431: FatFS exFAT is broken.

iabdalkader, 2019-01-26:
Hi, any interest in updating FatFS to the latest ? It's 2 years old now, and later revisions fix critical bugs (especially the broken exFAT implementation). I could work on a patch, but I will definitely need some help. Here's a list of changes/bug fixes since 0.12b http://elm-chan.org/fsw/ff/updates.txt

dpgeorge, 2019-01-26:
Yes this would be good to do.  The work should be done in the repo https://github.com/micropython/oofatfs

BASM, 2019-02-11:
> Yes this would be good to do. The work should be done in the repo https://github.com/micropython/oofatfs

What do you think about SPI FS? Why Fat?

dpgeorge, 2019-02-13:
> What do you think about SPI FS? Why Fat?

There's no problem to support multiple FS types.  FAT FS was chosen because it then allows a PC to easily access the filesystem, eg over USB mass storage.  And see #3847 for LittleFS support.

dpgeorge, 2019-02-21:
The oofatfs repo was updated to FatFS R0.13c.  But merging it here is currently blocked due to a suspected bug in the new FatFS.

closed 2019-02-21 by iabdalkader