micropython/micropython #4434: ESP32, V1.10.19: Wrong baudrate on UART at device with 26Mhz crystal

robert-hh, 2019-01-28:
The new build with version 1.10.19 run wit the wrong baud rate at ESP32 modules with a 26 MHz crystal, instead of 115200 baud, The initial debug messages are shown with the proper baud rate, but then the device switches to 74880 Baud (115200 * 26 / 40). That worked still fine in yesterday's version.

dpgeorge, 2019-01-28:
Ok, it looks like it needs auto-xtal detection.  Tyr adding `CONFIG_ESP32_XTAL_FREQ_AUTO=y ` to sdkconfig.

robert-hh, 2019-01-28:
There are quite a few of them: Which sdkconfig?

robert-hh, 2019-01-28:
Found a place that change the picture -> worse. Now I get this message:
```
W (56) rtc_clk_init: Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 26 MHz.
��n����rl���l��l�8�����|��Ă��
9��ln��0��                      �|~������<��
             b
```
More gibberish follows. And the baud rate is still 74880 baud.

robert-hh, 2019-01-28:
Putting that into:
..../esp32/boards/sdkconfig
worked.

dpgeorge, 2019-01-28:
Thanks for testing the fix.  It's now applied in f431795caf99ef49a5a1010063146ba25f524548

closed 2019-01-28 by dpgeorge