micropython/micropython #4435: [STM32H743] ADC fail

Meekdai, 2019-01-28:
Using the latest version of the main branch and run the script.
```
>>> import pyb
>>> adc = pyb.ADC(pyb.Pin.board.PA4)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: pin A4 does not have ADC capabilities
```
https://github.com/micropython/micropython/blob/9ac9aa989c89a1d6310160dd408820d1164d4dd7/ports/stm32/adc.c#L396-L399
Because H7 uses ADC3
https://github.com/micropython/micropython/blob/9ac9aa989c89a1d6310160dd408820d1164d4dd7/ports/stm32/adc.c#L53-L58

dpgeorge, 2019-01-29:
Thanks for the report.  It looks like there are a few other issues: the ADC capabilities need to be defined in the boards/stm32h743_af.csv file, and the use of the pin_adc1 array should be changed to pin_adc3 for the H7.

orelmi, 2019-02-04:
@dpgeorge, please could you explain how to implement ADC capabilities on STM32H743ZI ?

dpgeorge, 2019-02-06:
I added basic support for H7 ADC on pins in 43a894fb4807ad9ec25569952c0dcd9fd5525a87 and b367c425e40f3792dba625a33a625e345cec723c

On the NUCLEO_H743ZI board only the following pins are supported at the moment: PF6-PF10 and PH2.

orelmi, 2019-02-09:
I use this patch to use not simultaneously ADC1/ADC2/ADC3 on NUCLEO_H743ZI

It's important to enable clock on ADC1/ADC2 to use it.

@dpgeorge any idea and is it possible to use simultaneously ADC1/ADC2/ADC3 ?

`
#if defined(STM32H7) 

#define STM32H7_ADC1_ENABLED (1)

#if defined(STM32H7_ADC1_ENABLED)
#define ADCx                    (ADC1)
#define PIN_ADC_MASK            PIN_ADC1
#define pin_adc_table           pin_adc1
#define ADCx_CLK_ENABLE         __HAL_RCC_ADC12_CLK_ENABLE
#endif

#if defined(STM32H7_ADC3_ENABLED)
#define ADCx                    (ADC3)
#define PIN_ADC_MASK            PIN_ADC3
#define pin_adc_table           pin_adc3
#define ADCx_CLK_ENABLE         __HAL_RCC_ADC3_CLK_ENABLE
#endif

#else

#define ADCx                    (ADC1)
#define PIN_ADC_MASK            PIN_ADC1
#define pin_adc_table           pin_adc1
#define ADCx_CLK_ENABLE         __HAL_RCC_ADC1_CLK_ENABLE
#endif
`

dpgeorge, 2021-07-02:
The issue here is partially addressed by ff7be31f264c359b6f11fcc2cec8489dfe4905bd, so now ADC12 is used by default for `pyb.ADC(pin)`.  The final step would be to allow the user to select which of ADC1/2/3 to use as an argument to `pyb.ADC()`.