micropython/micropython #4436: ESP32: Non-volatile storage support

dmartauz, 2019-01-28:
This feature would be great for storage of configuration parameters, flags and other data in between power cycles or during deep-sleep while avoiding file system corruption which might occur during power outage in case data would stored in the file system (currently the only option).

ESP-IDF provides an NVS library:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_flash.html

dpgeorge, 2019-01-29:
> while avoiding file system corruption which might occur during power outage in case data would stored in the file system (currently the only option).

An alternative filesystem which is robust against power outages during write is LittleFS, and this is due to be integrated soon.  See #3847 for more details.

> ESP-IDF provides NVS library:

I guess it could be exposed as a module, but it's more Python-like, and portable to other platforms, to use files for this purpose.

dmartauz, 2019-02-28:
@dpgeorge 
A question regarding storage of data in the current implementation of file system in ESP32 port - is it suitable for frequent rewrites of data? I mean is there some wear-leveling mechanism implemented? My goal is to store data in between deep-sleep wake-ups.

Lisa999, 2019-04-01:
I have NVS working after making some mods to the source code. However, since @dpgeorge is quite slow with merging PR's i'm not sure whether to put the effort in making a NVS PR....

dmartauz, 2019-04-02:
Yes, please make a PR, it is worth it.
Which data types are currently supported by your implementation?

Lisa999, 2019-04-13:
All data types are support AND i’m using it in my own product. However, as long @dpgeorge is not merging PR #3576 i’m not putting in any effort in a PR that never gets merged....
So here you go @dpgeorge , all or nothing: merge #3576 and i’ll make a PR for the NVS part OR nothing....

dpgeorge, 2021-02-19:
NVS support was added in c10d431819f9f7095c3573c98f01c98526a2cb0b

closed 2021-02-19 by dpgeorge