micropython/micropython #4437: STM32 UART buffer silent overrun

TravisJoe, 2019-01-28:
Often this issue can be overlooked by making the buffer bigger than worst case scenarios, flow control, or just good planning. However I dislike not knowing a buffer is full and I started to lose data. With at least some hardware buffers the oldest data is pushed out so you retain at least the latest information, but I am finding that the UART buffer just stops filling the buffer once full. 

I understand there is probably a lot more to this than what I see on the surface, but it would be good to at least clarify why the buffer is not circular, and why an exception is not thrown.

dpgeorge, 2019-01-29:
> but it would be good to at least clarify why the buffer is not circular, and why an exception is not thrown

The UART RX buffer is circular, but it has the behaviour that when it's full it stops accepting new chars into it.  When flow control is enabled (CTS/RTS) this means that CTS signals to the other side to stop sending.

If there's an overrun of incoming data then whether it discards new or old data really depends on the application using it (and maybe this could be made run-time configurable).  But it would definitely be an improvement if it at least let you know that an overrun occurred.

TravisJoe, 2019-02-12:
At least on some of the other communication protocols it throws exceptions when there is an issue and sometimes at first that is annoying, but at least it is obvious. 

In a perfect world an exception would be thrown, but that might break too much functionality. Maybe similar to CAN you can call a function that returns buffer related information so you can at least poll its status.

peterhinch, 2019-02-12:
Optional callbacks when the rxbuf is "nearly full" and "nearly empty" would enable the user to implement various strategies including RTS/CTS or XON/XOFF.

TravisJoe, 2019-02-12:
I like that approach which is related to a similar ticket regarding software RTS/CTS. A callback would allow for a reasonable in application management of this, which is basically what is being done now with the client. 

So optional callback and a function to get uart/buffer status would more than solve my problems.