micropython/micropython #4438: UART RTS/CTS software flow control

TravisJoe, 2019-01-28:
This was brought up by somebody I am working with that has several UARTS implemented that have large packets of information and network communication. This causes potential for blocking functionality and has lead to buffer overflows on the UART. Flow control helps but not all UARTs have flow control and I was asked if that flow control could be done low level in Micropython. That sounded like an interesting enough idea to bring up here.

dpgeorge, 2019-01-29:
> Flow control helps but not all UARTs have flow control and I was asked if that flow control could be done low level in Micropython

I guess by the title that you mean to implement RTS/CTS in software using 2 extra GPIO lines?  As opposed to in-band flow control like Xon/Xoff?  Yes, I guess that could be done.

TravisJoe, 2019-02-12:
The need somewhat stems from this issue #4437 , which to me is more important. 

I do see some value in software controlled user defined RTS/CTS, but this stems the client designing a board and put higher bandwidth communication on one of the UART that does not have hardware RTS/CTS, and they do not want to do a redesign. Software RTS/CTS would almost be a work around.

I would not be hurt if this was closed.