micropython/micropython #4442: ESP32 Devkit v1 board + OV7725 Camera

sunbearc22, 2019-01-29:
May I know the uPython modules, classes, methods, functions that I can use to setup a ESP32 Devkit v1 board with a [OV7725 Camera](https://www.ovt.com/sensors/OV7725)? 

I came across the work by [Ivan Grokhotkov](https://github.com/igrr/esp32-cam-demo) who used an  ESP32 Devkit v2 board and the ESP-IDF (Espressif IoT Development Framework) (not uPython). Also, there is another work by BITLUNI'S LAB using a  ESP32 Devkit v1 board but with a OV7670 camera and Arduino.

Any advice/direction on using uPython to achieve this objective is appreciated. Thank you.

dpgeorge, 2019-01-30:
The esp32 port does not currently support the camera interface.  To get it working would require integrating the low-level C code, then exposing that to Python via a module.

sunbearc22, 2019-01-31:
Thank you. 

I found [OpenMV Cam H7](https://www.kickstarter.com/projects/1798207217/openmv-cam-h7-machine-vision-w-micropython) consist of a OV7725 and STM32H743VIT6. 

Their Github repo contains three low level C codes for the OV7725 ([OV7725.c](https://github.com/openmv/openmv/blob/master/src/omv/ov7725.c), [OV7725.h](https://github.com/openmv/openmv/blob/master/src/omv/ov7725.h), [OV7725_regs.h](https://github.com/openmv/openmv/blob/master/src/omv/ov7725_regs.h)). Also they have some [uPy modules](https://github.com/openmv/openmv/tree/master/src/omv/py) like sensor, image, time for user implementation. It is customized for STM32H743VIT6. Can you advice me on how to customize them for  ESP32 Devkit v1 and uPy?

dpgeorge, 2019-01-31:
> Can you advice me on how to customize them for ESP32 Devkit v1 and uPy?

Sorry but this is out of scope of what help can be provided here.  Using the OpenMV Cam code does look like a promising direction to take, and the https://forum.micropython.org would be a better place to ask for advice about integrating it.

closed 2019-01-31 by dpgeorge

sunbearc22, 2019-01-31:
Thank you for introducing me the micropython forum. Will use that. :)