micropython/micropython #4443: esp8266 xxxsleep(): Seem not to work as expected in the most recemnt commit

robert-hh, 2019-01-30:


dpgeorge, 2019-01-30:
> The time argument works, but the device does not recover well.

Did you connect IO16 to RST?

> machine.lightsleep(): functional OK, but the current consumption increases form ~25mA to ~75mA

I can't reproduce this.  With and without STA and/or AP active, power consumption on an Adafruit Feather HUZZAH reduces from 32mA to 29mA.

robert-hh, 2019-01-30:
Sorry. I seemed to be confused. ESP32 is fine. All trouble is with a an ESP8266.
ESP8266 Wemos D1 Mini: 
deepsleep: With GPIO16 connected to RST the device recovers well
lightsleep: Current increases from ~25mA to ~75 mA.
ESP32 Wemos D1 Pro. and Sparkfun ESP32 Thing
lightsleep: USB supply current decreases from ~45mA to 12 mA and ~8mA, which should mostly be caused by the USB/UART bridge.

robert-hh, 2019-01-30:
In my test I had the network interface on in STA mode. With the interfaces off, the current drops from ~25mA to 21 mA. So the difference is the same. It is just strange that the current saving method you implemented recently with WiFi enabled is not active when switching to light sleep.

dpgeorge, 2019-01-30:
> In my test I had the network interface on in STA mode.
> ...
> It is just strange that the current saving method you implemented recently with WiFi enabled is not active when switching to light sleep.

I managed to reproduce the problem you describe by activating STA and connecting to an AP with it.  Then lightsleep doesn't allow the WiFi DTIM sleeping to work correctly and current goes up.  This should now be fixed by 3ff3e968656cd01144c0b92e2f0cf4195740255d

Thanks for testing these things!

closed 2019-01-30 by dpgeorge