micropython/micropython #4447: build esp32 port including wiznet5k / w5500 driver

balku, 2019-01-30:
Hi,
like to build a esp32 port that include the W5500 driver. How can I do this? 
Thanks
balku

dmartauz, 2019-01-30:
I am also very interested in this. Having Ethernet support implemented via SPI adapter with the bus shared with other peripherals would be interesting alternative to RMII using built-in Ethernet MAC.

dpgeorge, 2019-01-31:
It is quite a lot of work to do this.  The W5500 can run in either managed mode (with TCP/IP stack on the w5500) or in MAC raw mode (TCP/IP stack provided by the host).  Integrating it in either of these modes is hard: managed mode requires a completely separate socket implementation (basically taken from the stm32 port) and it wouldn't work seamlessly with the wifi; and MAC raw mode requires hooking it in to the low lwIP layers provided by the ESP IDF.  Probably the latter is the better approach.

Such support (in either form) is not on the roadmap for esp32 and would need to be contributed.  In the meantime it supports Ethernet via other RMII adapters.

balku, 2019-01-31:
Hi Damien,
many thanks for your answer. Which RMII adapters would you recommend for a esp32 port of MP?
Thanks balku

dpgeorge, 2019-01-31:
> Which RMII adapters would you recommend for a esp32 port of MP?

I haven't tried any myself.  For previous discussion see https://github.com/micropython/micropython-esp32/issues/172 and https://github.com/micropython/micropython-esp32/pull/187

imliubo, 2020-12-08:
Hi, I saw ESP-IDF now is support W5500 in this commit: [eth: support W5500 in network examples](https://github.com/espressif/esp-idf/commit/aea901f014ac05c41664dd1e3ba9bbd2301a0aba).

So micropython is also can support it?

kenjican, 2020-12-29:
Bump! we do need support for W5500. I've tried LAN8720 with Wrover , but not luck. The network.LAN need GPIO 17 ,GPIO16 but these two pins are occupied by spiram. and GPIO0 not OUT mode.

    And RMII consumes more than 10 pins...it's a little bit wasteful.

newbie-gk, 2021-03-06:
Is this WIZNET5K available in the esp32 port yet?

tmecha, 2021-09-17:
Same: W5500 for ESP32 via WIZNET5k option? Can't tell that this issue actually got resolved.

beyonlo, 2021-11-24:
Hello. 

I'm interested to use the W5500 on the ESP32, specifically in the ESP32-S2 and ESP32-S3. Anyone know if actually the W5500 is supported by MicroPython port for the ESP32?

Thank you.

Ayyoubzadeh, 2022-09-28:
Hello,
Check this repo [https://github.com/Ayyoubzadeh/ESP32-Wiznet-W500-Micropython](https://github.com/Ayyoubzadeh/ESP32-Wiznet-W500-Micropython)

beyonlo, 2022-09-28:
@Ayyoubzadeh Hello!

That's great. Unfortunately I need `SSL` on driver and better performance, [like as a low level driver](https://github.com/micropython/micropython/pull/8540) (not a Python implementation)!  

I was starting as well using Python driver (from CircuitPython WizNet driver). Of course your driver is very more dedicated and more tested for MicroPython - excellent.

@andrewleech  has implemented is this [PR](https://github.com/micropython/micropython/pull/8540) the Wiznet W5100S/W5500 for the rp2040 and that was amazing, works with SSL and very good throughput/performance. The @liudr [started a interest](https://github.com/micropython/micropython/pull/8540#issuecomment-1153237218) to use that work done from @andrewleech to support the Wiznet on the ESP32 as well, but I don't know if that was really continued. I'm talking this story for you because maybe you can follow the same approach if you has interest in to continue that :)

Thank you very much!