micropython/micropython #4452: ESP32-WROVER deep-sleep current too high (85uA)

dmartauz, 2019-01-31:
Tested with ESP32-WROVER module with ESP32-D0WDQ6, 32Mbits PSRAM, 32Mbits SPI flash. Module is soldered on custom board. Only 10kOhm EN pin pull-up is soldered on the board except of the module.
Tested with both esp32-20190125-v1.10 and esp32spiram-20190125-v1.10.

Screenshot from Otii Arc tool:
https://drive.google.com/open?id=12WQP10Gq0hINqLwKueF2FvWOvxZWaf0L

dmartauz, 2019-01-31:
might be related: https://www.esp32.com/viewtopic.php?t=3634

dmartauz, 2019-02-06:
@dpgeorge Is it posible to expose above mentioned pin isolate function in MicroPython API?

dpgeorge, 2019-02-07:
> Is it posible to expose above mentioned pin isolate function in MicroPython API?

Yes I guess it makes sense, something like `esp32.rtc_gpio_isolate(pin)`.  Then it would be up to the user to know when to use it, and possibly create a custom `deepsleep()` wrapper function (around `machine.deepsleep()`) that isolates the correct pins give the board it's running on.

dmartauz, 2019-02-09:
I agree, this is the best approach.

dmartauz, 2019-02-27:
@dpgeorge Any time estimation for implementation of this feature? I don't want to push you I am just asking.

dpgeorge, 2019-02-28:
Looking closer at how it works, `rtc_gpio_isolate()` is just a helper function that calls other functions to disable pull ups and set the GPIO pin in hold mode.  So all the esp32 port is missing is the ability to enable the hold mode on a GPIO pin.  I've make a test to add such support in #4561.  @dmartauz can you please test this new code?

dmartauz, 2019-02-28:
@dpgeorge I encountered this error, what am I doing wrong?
`
MicroPython v1.9.3-1571-g1cc3f06 on 2019-02-28; ESP32 module with ESP32
Type "help()" for more information.
>>> import machine
>>> pin12=machine.Pin(12,mode=machine.Pin.IN,pull=None)
>>> import esp32
>>> esp32.gpio_hold(pin12)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 2 positional arguments but 1 were given

`

dpgeorge, 2019-02-28:
Thanks for testing!  Yes, sorry, I wrote it incorrectly, how to use gpio_hold, it does need a second argument.

dmartauz, 2019-02-28:
No problem, I am glad that the issue is solved.

dpgeorge, 2019-03-26:
This was eventually implemented with `Pin.PULL_HOLD`.

closed 2019-03-26 by dpgeorge

dmartauz, 2020-04-28:
Short remark: Pin 12 is probably connected differently on ESP32-WROVER-IB (16MB). When I tried the code I was running before on original ESP32-WROVER including line with Pin.PULL_HOLD for pin 12 before calling machine.deepsleep() I experienced higher deep sleep current with the new module. Removing the Pin.PULL_HOLD line helped to reduce the deep sleep current to expected level.