micropython/micropython #4453: qstr generation may use gcc instead of $(CPP)

pmp-p, 2019-01-31:
at least i think so, as  the step
`GEN build/genhdr/qstr.i.last`
is not very clear to me.

also maybe here too https://github.com/micropython/micropython/blob/67689bfd7e28df8d4fa23c8f25065aeb0b9d9658/py/mkrules.mk#L56

dhylands, 2019-01-31:
CFLAGS also includes -D options which directly affect code generation and other options (like -mthumb) which indirectly cause defines to be created which if not present would generate different pre-processed output.

dhylands, 2019-01-31:
It seems CPP is already defined to be $(CC) -E. At least that's the default for my make (GNU make v4.1 running under ubuntu).

What is it that you're trying to do and what specific errors are you running into?

pmp-p, 2019-01-31:
i'm cross compiling with a non standard toolchain where the prepropessor is gcc and won't accept cc specific options ( it's a python wrapper from emscripten around clang known as emcc ).

pmp-p, 2019-01-31:
well thanks, COPT is ok to act as "C OPTIONS" :) and fixed my preprocessor problems. but a bug can hide another and in fact frozen object is twice on the final link nothing to do with CPP ( i hope ! )

closed 2019-01-31 by pmp-p