micropython/micropython #4455: mpy-cross broken on macOS Mohave

annejan, 2019-02-01:
It compiles fine, but won't run
```
./mpy-cross 
dyld: malformed mach-o image: missing LC_DYSYMTAB
Abort trap: 6
```
I have tried not stripping by removing `,-dead_strip` from the Makefile to no avail . .
```
 ./mpy-cross 
Killed: 9
```

NB: This is not the same issue as #2941 since this is on the current master.

tannewt, 2019-02-02:
Try removing binutils that brew installs. I think that was the issue for me.

dpgeorge, 2019-02-03:
I'd suggest this could be related to #3637 : maybe the ABI is different and the NLR code doesn't work.

andrewleech, 2019-02-05:
Out of interest, would you mind trying the latest master osx binary from my CI build of mpy-cross? 
osx binary:
https://gitlab.com/alelec/mpy_cross/-/jobs/156257348/artifacts/file/mpy-cross

pip install-able copy: 
https://gitlab.com/alelec/mpy_cross/-/jobs/155326779/artifacts/file/mpy_cross-1.11.dev45+g67689bfd7-py2.py3-none-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl

I don't have access to a new enough mac to test these on myself.

Anton-2, 2019-02-06:
Tested with the latest Mojave (macOS 10.14.3), and up to date command line tools :

- The version from your gitlab works ...
 and 
- The version build from a fresh git clone works too.

andrewleech, 2019-02-06:
Great to hear on both counts, thanks!

mcauser, 2019-04-04:
I ran into this one today.
`brew uninstall binutils` fixed it for me, macOS High Sierra 10.13.6

dpgeorge, 2019-04-08:
@andrewleech it sounds like you manage to fix this, how did you do it?

@mcauser are you able to build and execute the unix port of MicroPython on the same machine that mpy-cross fails (with binutils still installed)?

andrewleech, 2019-04-08:
I build my osx executables with crossbuild docker images, on linux host: https://hub.docker.com/r/multiarch/crossbuild

Gitlab make the CI very easy: https://gitlab.com/alelec/mpy_cross/blob/master/.gitlab-ci.yml
Ok, that probably looks a bit complex, but that's building it for a lot of platforms from the one configuration and bundling it in wheels for easy distribution. That reminds me, I need to fix my release process there for the 1.10 release still.

dpgeorge, 2019-04-08:
> I build my osx executables with crossbuild docker images

I see, so that's probably building with a good binutils.  So the real reason for the problem is still unknown.

hyhkjiy, 2019-06-01:
You need rebuild mpy-cross.
```
cd micropython/mpy-cross
rm mpy-cross
```
Run `printenv | grep FLAGS` before build mpy-cross, run `runset`(like: `unset LDFLAGS CPPFLAGS` ) if have something. 

Now build mpy-cross:
```
make
```
This is very effective on my computer.

jonnor, 2023-07-30:
It seems this issue has been resolved?

closed 2023-07-31 by mattytrentini